sn4k3/UVtools #527: [BUG] v3.5.5 - Repair Suction Cup damage object

antwal, 2022-07-28:
## System
 - OS: MacOS Catalina 10.15.7
 - Slicer: PrusaSlicer 2.4.2
 - Printer: ELEGOO Saturn
 - UVtools version: 3.5.5

```
UVtools v3.5.5 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Mon Apr 18 21:50:40 PDT 2022; root:xnu-6153.141.62~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz
Processor cores: 16
Memory RAM: 16.47 / 64.95 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.16
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 200% (Primary)
    WA: 2560 x 1417    UA: 1280 x 708
2: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708
3: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708

Path:       /private/var/folders/l0/9gshyl5j3h3d2ps8dtcstv8w0000gn/T/AppTranslocation/0BCE309C-D140-4383-AD88-5A9F70E56D2B/d/UVtools 3.5.5.app/Contents/MacOS/
Executable: /private/var/folders/l0/9gshyl5j3h3d2ps8dtcstv8w0000gn/T/AppTranslocation/0BCE309C-D140-4383-AD88-5A9F70E56D2B/d/UVtools 3.5.5.app/Contents/MacOS/UVtools
Loaded file: ryuk_SubTool_3_Fix_0.05mm_5h45m.sl1 [Version: 0] [Class: SL1File]

```

## Describe the bug
This is a problem that has always existed and has never been fixed, I include 2 sample videos between the 3.3.2 version I use and the latest version 3.5.5 released.

To solve the suction cup problem, I add holes to the objects where possible and position them so that they have 1/2 holes above and 1/2 holes below the object, avoiding to repair the problem from uvtools.

## To Reproduce
Steps to reproduce the behavior:
Open any file that has a suction cup and try to repair it

## Video Test
[3.5.5 Test](https://youtu.be/RA2LWcnUQX0)
[3.3.2 Test](https://youtu.be/C0c6hXIANHk)

sn4k3, 2022-07-29:
It's not a bug, it's how UVtools works to attempt fix the suction cup. A drill hole will always damage model since you are removing pixels that should been there in first place.

Suction cups should always be fixed on slicer when possible, that's why the default is not to repair them.
The way UVtools repair works it is very primitive, it just drill a vertical hole in Z axis at the Centroid of the mass. So it's a blind drill and doesn't check for best position nor try to adjust axis to make perpendicular hole to the wall. This would require a ton of analysis and not so easy to do. That way it just do the hole at it centroid and hope for the best. Depending on model and orientation some might get luck others don't. 

I always recommend do the drain holes on slicer, this way you can choose the best position and orientation, things that UVtools can't do and probally will never do at such degree.

antwal, 2022-08-01:
ok i got it, so i don't consider it as bugs.

Wouldn't it be possible to apply a pattern to the suction cap in order to reduce the area without applying a vertical hole to the object?

sn4k3, 2022-08-01:
A random pattern in XY will have negative results. Without knowing the best layer and position to place the hole there's no way to optimize. Such analysis would be more time consuption than check for the issues itself.

All you can do is place the hole yourself, under pixel editor -> Drain hole. That way you can choose position but stills a vertical hole

antwal, 2022-08-02:
OK! I got it

closed 2022-08-02 by antwal