sn4k3/UVtools #530: [BUG] Anycubic Mono X 6k - PWMB Problems

antwal, 2022-08-05:
## System
 - OS: macOS Catalina
 - Slicer: UVTools
 - Printer: Anycubic Mono X 6K
 - UVtools version: 3.5.6

```
UVtools v3.5.6 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Mon Apr 18 21:50:40 PDT 2022; root:xnu-6153.141.62~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i5-4570S CPU @ 2.90GHz
Processor cores: 4
Memory RAM: 8.98 / 14.58 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.17
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1057    UA: 1920 x 1057
2: 1366 x 768 @ 100%
    WA: 1366 x 745    UA: 1366 x 745

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]

```

## Describe the bug
a size error occurs during the file conversion then generates this problem on the printout (see screenshot)

## To Reproduce
Steps to reproduce the behavior:
1. Open UVTools
2. Menu Calibration -> Elephant foot
3. File -> Convert to -> Photon Mono 6K
4. Version 1 or 515 or 516

## Screenshots
[image]

[image]

[image]

sn4k3, 2022-08-05:
You can't use the demo file to generate tests or whatever to your printer.
You must use a dummy previous sliced file for your printer.

Demo file is just there to able to explore UVtools without a file but you ALWAYS need to open your own

antwal, 2022-08-12:
I understand, but I also have other problems with the files generated on UVTools, I am attaching the problems and PWMB files for possible debugging

[PWMB Files](https://ufile.io/f/vwtiu)

Version 1 and 515 the Z axis does not go up

With version 516 it works but the prints fail, I also attach the screenshots of the printed pieces.

[image]

[image]

[image]

sn4k3, 2022-08-12:
> Version 1 and 515 the Z axis does not go up. With version 516 it works.

Newest Anycubic machines uses only v516, other versions will fail.
With my PS profiles it output the correct version.

> but the prints fail

Can't see where this is realated with UVtools...
Either your supports are failling or you are one of the unlucky guys to get the machine with bad board that shift image when AA is ON. Try to print same file but with AA disabled (You must slice on Photonworkshop and disable AA)

antwal, 2022-08-12:
> > Version 1 and 515 the Z axis does not go up. With version 516 it works.
> 
> Newest Anycubic machines uses only v516, other versions will fail. With my PS profiles it output the correct version.
> 

i installed the profile from uvtools it should be correct right?

> > but the prints fail
> 
> Can't see where this is realated with UVtools... Either your supports are failling or you are one of the unlucky guys to get the machine with bad board that shift image when AA is ON. Try to print same file but with AA disabled (You must slice on Photonworkshop and disable AA)

ok I also try this thing

with photon workspace 2.1.29 and lycheeslicer the printing is successful without problems, I also tried chitubox free and pro generates problems similar to prusa + uvtools

antwal, 2022-08-12:
ok can i disable it for testing to check if it works, on prusa? thx

sn4k3, 2022-08-12:
On prusa you can disable AA setting gamma to 0 BUT you also need to disable on UVtools which you need to use the terminal console for that (No direct way on UI) that why i adviced to try with photon workshop