sn4k3/UVtools #533: [BUG] 'wait time before cure' suggestion doesn't seem to apply properly

dicethrow, 2022-08-08:
When I import a file, I am unable to ever make the 'wait time before cure' behave as expected.

Note that I have never been able to make the suggestions 'apply' box appear non-greyed-out; so this issue could be part of the suggestions mechanism, or part of the 'wait time before cure' functionality.

## System
 - OS: Ubuntu 22.04
 - Slicer: Anycubic Photon Workshop 2.1.29
 - Printer: Anycubic Photon Mono X 
 - UVtools version: 3.5.6

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

UVtools v3.5.6 X64
Operative system: Linux 5.17.5-051705-generic #202204271406 SMP PREEMPT Wed Apr 27 16:11:45 UTC 2022 X64
Processor: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx
Processor cores: 8
Memory RAM: 11.81 / 29.33 GB
Runtime: ubuntu.22.04-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.17
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 125% (Primary)
    WA: 2560 x 1413    UA: 2048 x 1130

Path:       /home/x/Documents/git_repos/forks/mechanics/tools/UVtools/__prebuilt/UVtools_linux-x64/
Executable: /home/x/Documents/git_repos/forks/mechanics/tools/UVtools/__prebuilt/UVtools_linux-x64/UVtools
Loaded file: Dragon_v2_0sec_offtime.pwmx [Version: 516] [Class: PhotonWorkshopFile]

```

## Describe the bug
The 'wait time before cure' suggestions functionality is never applied, even if the file parameters should trigger the suggestion to be applied. e.g. the screenshot shows a file with 0sec wait-time-before-exposure, set in my slicer; and the suggestion settings contains the settings that the suggestion will be applied if the file has different wait-time-before-exposure settings; but as you can see, that never seems to be recognised.

## To Reproduce
Steps to reproduce the behavior:
1. Open any file
2. Click the 'suggestions' tab
3. Set 'wait time before cure' settings as shown in the screenshot, then close the settings window

## Expected behavior
I expect a pop-up checkbox will appear for the 'wait time before cure' suggestion, that I can then select 'apply' for. I then expect the entry in 'Applied Suggestions', and the numbers in the statusbar, to contain the numbers that I specified in the settings.

Instead, no such checkbox appears, and 'apply' is always greyed out. The entry in 'Applied Suggestions' and the statusbar contains the wait time embedded in the source file with no modifications made.

## Screenshots
main window, with a file open
[image]

the current settings for my suggestions
[image]

## Files
Source file has copyright. However any file I use, including the test file, has this behavior.

## Additional context
Following this amazing blog 
https://blog.honzamrazek.cz/2022/01/prints-not-sticking-to-the-build-plate-layer-separation-rough-surface-on-a-resin-printer-resin-viscosity-the-common-denominator

sn4k3, 2022-08-08:
Anycubic firmware is bad. They only have 1 global wait time property. That means you can't use this tool/blog advice, because theres is no way to define bottom and normal layers wait time, only one time that affect every layer. You can thank Anycubic for doing proper firmware.

dicethrow, 2022-08-08:
ah! ok, thanks for the quick advice. In that case I'll stick with using long waits for all layers.

closed 2022-08-08 by dicethrow