sn4k3/UVtools #537: ctbv4 encryption message not displaying correctly

frozenowni, 2022-08-21:
Hi,
I recently got a saturn 2 and therefore started using ctbv4 files...

The informative message upon opening was not displayed correctly on my pc. I use a 15" screen in full hd 16:9.
it was only saying "informative message ..."
I only found out about the rest when trying to select all by triple-clicking...

Can you maybe enlarge the message window so all users see the background info regarding the ctbv4 fileformat.

Thanks for the great tool btw!

BR Jonas

sn4k3, 2022-08-21:
Also reported here: https://www.facebook.com/groups/uvtools/posts/740410317068474

This is a bug with message box library (external from UVtools).
Will be fixed on next release with PR: https://github.com/AvaloniaCommunity/MessageBox.Avalonia/pull/113

frozenowni, 2022-08-21:
Ok great thx

sn4k3, 2022-08-22:
Fixed

closed 2022-08-22 by sn4k3