sn4k3/UVtools #538: [Crash] System.IndexOutOfRangeException: Index was outside the bounds of the array.

Inseckto, 2022-08-21:
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.FileFormats.FileFormat.get_Item(UInt32 index) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.Core/FileFormats/FileFormat.cs:line 3025
   at UVtools.WPF.MainWindow.ShowLayer() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.LayerPreview.cs:line 849
   at UVtools.WPF.MainWindow.RunOperation(Operation baseOperation) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.axaml.cs:line 2146
   at UVtools.WPF.MainWindow.ShowRunOperation(Type type, Operation loadOperation) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.axaml.cs:line 2019
   at UVtools.WPF.MainWindow.OnClickRepairIssues() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.Issues.cs:line 582
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Native.PlatformThreadingInterface.SignaledCallback.Signaled(Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 39
   at Avalonia.Native.Interop.Impl.__MicroComIAvnSignaledCallbackVTable.Signaled(Void* this, Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/Interop.Generated.cs:line 4336
--- End of stack trace from previous location ---
   at Avalonia.Native.PlatformThreadingInterface.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 90
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/Program.cs:line 70
Category: Application

# System
UVtools v3.6.1 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Wed Aug 10 14:25:27 PDT 2022; root:xnu-8020.141.5~2/RELEASE_X86_64 X64
Processor: 
Processor cores: 16
Memory RAM: 6.79 / 14.01 GB
Runtime: osx.12-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools

Machine date time: 08/21/2022 15:54:23
    UTC date time: 08/21/2022 20:54:23

sn4k3, 2022-08-21:
Can attach the problematic file?

Inseckto, 2022-08-21:
Sure! [Exposure Tester 7 v2_Resi.ctb.zip](https://github.com/sn4k3/UVtools/files/9389949/Exposure.Tester.7.v2_Resi.ctb.zip)

sn4k3, 2022-08-21:
That file works for me on v3.6.1

Inseckto, 2022-08-21:
For me it crashes every time I try to auto-remove the detected empty layers (v3.6.1)
[image]
.

sn4k3, 2022-08-21:
Ah yes, because files can't contain 0 layers.
UVtools is not prepared for it and always expect to have 1 layer.

Inseckto, 2022-08-21:
I see, maybe the file is damaged, as is not supposed to have empty layers only. Thanks!

closed 2022-08-21 by Inseckto

sn4k3, 2022-08-21:
That file contain no valid layers. It will print nothing if you try.