sn4k3/UVtools #540: [BUG]

roncromberge, 2022-08-24:
# bug report. Error at running issues at the file open stage

## System
 - OS: macOS 12.5.1 (21G83) Monteray
 - Chip:  Apple M1 Pro
 
 - Slicer: Lychee 10.0.11 and Chitubox basic V1.9.3
 - Printer: Elegoo Saturn Firmware ctb V4 compatible
 - UVtools version: UVtools v3.6.2 X64 Bug started @ version 3.5.9 and onwards

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

UVtools v3.6.2 X64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Wed Aug 10 14:28:23 PDT 2022; root:xnu-8020.141.5~2/RELEASE_ARM64_T6000 X64
Processor: Apple M1 Pro
Processor cores: 10
Memory RAM: 7.34 / 12.69 GB
Runtime: osx.12-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1800 x 1169 @ 200% (Primary)
    WA: 1800 x 1084    UA: 900 x 542

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Phrozen_XP_Finder_V2.ctb [Version: 4] [Class: ChituboxFile]

## Describe the bug
When I load uvtools stand alone nothing happened. when I load a self sliced ctb file the error message appear in the "issues" phase.

```
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.Managers.IssueManager.DetectIssues(IslandDetectionConfiguration islandConfig, OverhangDetectionConfiguration overhangConfig, ResinTrapDetectionConfiguration resinTrapConfig, TouchingBoundDetectionConfiguration touchBoundConfig, PrintHeightDetectionConfiguration printHeightConfig, Boolean emptyLayersConfig, OperationProgress progress)
   at UVtools.WPF.MainWindow.<>c__DisplayClass212_0.<ComputeIssues>b__0() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.Issues.cs:line 620
```

## To Reproduce
Steps to reproduce the behavior:
1. Start uvtools from lychee And error message appear direct after loading.
2. Start uvtools from desktop icon. The error appears after open a cab file

3. See error
```
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.Managers.IssueManager.DetectIssues(IslandDetectionConfiguration islandConfig, OverhangDetectionConfiguration overhangConfig, ResinTrapDetectionConfiguration resinTrapConfig, TouchingBoundDetectionConfiguration touchBoundConfig, PrintHeightDetectionConfiguration printHeightConfig, Boolean emptyLayersConfig, OperationProgress progress)
   at UVtools.WPF.MainWindow.<>c__DisplayClass212_0.<ComputeIssues>b__0() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.Issues.cs:line 620
```
# extra info:
I still use the intel version... Because I still can't install and run the Arm version ;-/

[affected_ctb_files.zip](https://github.com/sn4k3/UVtools/files/9415245/affected_ctb_files.zip)

sn4k3, 2022-08-24:
Both files are working for me. No error after detect issues.

[image]

Have you changed any setting on UVtools or using defaults?

roncromberge, 2022-08-24:
Thanks for your fast response 👍🏼

As far as I know I did not change anything. But you never know when you trying things out 😀

I will clean all uvtools files on my MacBook and install it al over again. Where are the config files kept?

Best regards,

Ron Cromberge.

sn4k3, 2022-08-24:
On settings you can reset all to defaults and save.

Settings are kept on Help -> Open settings folder

roncromberge, 2022-08-24:
Hi,

Succes!

I cleaned all uvtools related directories and uvtools works like a charm.

Thanks for your fast response. And keep on the good work.

Best regards,
Ron Cromberge

closed 2022-08-24 by sn4k3