sn4k3/UVtools #541: [BUG]  PhotonWorkshopFile (pws) layers not decoded correctly

SG-O, 2022-08-26:
## System
 - OS: Linux Mint 20.3
 - Slicer: Chitubox 1.9.3
 - Printer: n/a
 - UVtools version: 3.6.2

UVtools v3.6.2 X64
Operative system: Linux 5.4.0-113-generic #127-Ubuntu SMP Wed May 18 14:30:56 UTC 2022 X64
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Processor cores: 12
Memory RAM: 14.05 / 15.49 GB
Runtime: linuxmint.20.3-x64
Framework: .NET 6.0.6
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1080    UA: 1920 x 1080

Path:       /tmp/.mount_UVtoolA9cFRJ/usr/bin/
Executable: /tmp/.mount_UVtoolA9cFRJ/usr/bin/UVtools
Loaded file: testFile.pws [Version: 516] [Class: PhotonWorkshopFile]

The bugs:
1. The layers are not decoded correctly.
2. The preview has the wrong color. Should be light blue on gray, is orange on gray.

[testFile.pws.zip](https://github.com/sn4k3/UVtools/files/9431871/testFile.pws.zip)

sn4k3, 2022-08-26:
The problem is with file, you name it .pws but is encoded to be pw0. If you rename to pw0 it will open fine.
Can this file print under your printer?
What is your printer?

SG-O, 2022-08-27:
I was creating test files for an App I'm working on when I came across this file that didn't fit what I expected. I tried opening it in UVTools and came across this error. I was using a custom printer profile in Chitubox. I can't print the file as I don't own a printer capable of printing this file.

Besides this the color for the preview is still not decoded correctly. Probably the red and blue channels are swapped.

sn4k3, 2022-08-27:
Chitubox don't goes well with custom printers. If you can't select the file flavor and you save as .pws but it encodes like pw0 then is a bug on thier software. 

About the preview, many users print files processed by UVtools and so far didn't get any complain about that.
Without proper evidence i will not blind change it. 
To test this out you need to slice a file on original slicer (PhotonWorkshop) process it with UVtools and save a copy to keep original. Load into printer and observe the previews. Does colors are different? If so that's the evidence i need to make the change.

closed 2022-08-30 by sn4k3