sn4k3/UVtools #545: [Crash] System.OverflowException: Value was either too large or too small for a Decimal.

federico9292, 2022-08-28:
System.OverflowException: Value was either too large or too small for a Decimal.
   at System.Number.ThrowOverflowException(TypeCode type)
   at System.Decimal.DecCalc.VarDecFromR4(Single input, DecCalc& result)
   at System.Decimal.op_Explicit(Single value)
   at UVtools.Core.FileFormats.FileFormat.RefreshPrintParametersModifiersValues() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 4017
   at UVtools.Core.Operations.OperationEditParameters.InitWithSlicerFile() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationEditParameters.cs:line 163
   at UVtools.Core.Operations.Operation.set_SlicerFile(FileFormat value) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\Operation.cs:line 80
   at UVtools.WPF.MainWindow.ProcessFile(String fileName, FileDecodeType fileDecodeType, UInt32 actualLayer) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1603
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application
File: 12MaxProThin_2.ctb  [Version: 4] [Class: ChituboxFile]

# System
UVtools v3.6.2 X64
Operative system: Microsoft Windows 10.0.22000 X64
Processor: Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz
Processor cores: 12
Memory RAM: 12.38 / 15.84 GB
Runtime: win10-x64
Framework: .NET 6.0.8
AvaloniaUI: 0.10.18
OpenCV: 4.5.5-237

Path:       C:\Users\feder\Downloads\UVtools_win-x64_v3.6.2\
Executable: C:\Users\feder\Downloads\UVtools_win-x64_v3.6.2\UVtools.exe

Machine date time: 08/29/2022 01:37:37
    UTC date time: 08/28/2022 23:37:37

sn4k3, 2022-08-29:
Looks like a bad value on file. Can you attach the file inside a zip?

federico9292, 2022-08-29:
I'm sorry I don't think I still have the bad file; I managed to change a
parameter (support density) by adding 1% to get it to reslice, and the
problem was gone. I thought the debug log was sufficient, my bad

sn4k3, 2022-08-29:
Debug log only tell where the problem happens, but miss the parameter and the value it was attempting to convert.
Without file i can't check whats wrong or how to fix it. Most likely a value that overflow. But support density is not used by UVtools, must be one of the Tool - Edit Print Parameters

federico9292, 2022-08-29:
I tried the pixel edit feature, but rolled back the edits, saved, then
tried to open it again later cause the printer wasn't recognizing the
file... maybe this helps?

sn4k3, 2022-08-29:
Nope... Not a clue, sorry :/

sn4k3, 2022-08-30:
Closing this, if you ever saw this problem again please attach file here

closed 2022-08-30 by sn4k3