sn4k3/UVtools #546: [BUG] Updating removing taksbar shortcut (WIN11)

hugoalves6, 2022-08-30:
## System
 - OS: Windows 11 
```

## Describe the bug
In all versions the same append, after installing the update (at least on Windows 11), the taskbar shortcut gets an error has if it was an empty shortcut.
I don't know if the developer knows this mini bug or not, but it's a bit pain, to go to installation folder and drag a new shortcut to the taskbar every time the user updates the program.

Best regards,
Thank you for your hard work.

sn4k3, 2022-08-30:
That's strange, but the cause is because everytime UVtools upgrades on Windows it get uninstalled first, that's because upgrading without install don't work so well. Since path is the same shortcut should survive but looks like not.
Maybe instead of task the UVtools.exe directly you can create a shortcut, link to UVtools.exe and pin to task?

hugoalves6, 2022-08-30:
I've tried that, but it does the same thing.

I think i didn't say it but i cannot drag the exe to the taskbar, i have to open the app, and then right click it and select add to taskbar, the shortcut doesn't allow to drag & drop to taskbar either.

After the update is done, i check the address of the taskbar shortcut, and its the same, everything in the proprieties is going to the correct file and folder, that is why I think it's a very weird behavior.

sn4k3, 2022-08-31:
Can't find a way to fix this with current installer. It looks like Win11 annoyance.
Also this wix don't support pin to taskbar or start without extra .exe packed with UVtools.

Your best bet is pin a shortcut that links cmd instead of UVtools and call UVtools from cmd command, eg:

`C:\Windows\System32\cmd.exe /c start "" "C:\Program Files\UVtools\UVtools.exe"`

[image]

hugoalves6, 2022-09-02:
Ok thx for that help, hope you find someway to fix it in future.

Best regards

closed 2022-09-02 by hugoalves6