sn4k3/UVtools #551: [BUG] Error because of locked file, causes next save to be corrupted.

l0rdbyte, 2022-09-14:
If a file is still open in another program (in my case .ctb in Chitubox), UVTools will give an error message. Saving again (after closing the offending file) will create a corrupt file.
Save as... a different filename seems to fix it. Haven't tried with saving a third time.
100% reproducable.

Project file & How to reproduce
Any file (only know of .ctb though) will work. Make sure it is open in another program (like chitubox). Saving will then give an error. Saving again will create a corrupt file. I can send one of those (one was a layer cake of the original with multiple layers of 100% fill, another just had a few layers and nothing else).
Looks like it's something with Filestream that's not properly closed.

Version of UVTools
3.6.3

Operating system
Windows 10 x64

Printer model
Phrozen Sonic Mini 8k

sn4k3, 2022-09-14:
I have a clue of what is happening (Full-encode flag is not getting reset in case of exception), but is a good cath.
However i don't understand why Chitubox opens ctb with write privilege and keep it stream open after load, as far i know you can't change any data on ctb when reopening on slicer? And so it should be open and release to read-only.

I will get into it

sn4k3, 2022-09-15:
fixed

closed 2022-09-15 by sn4k3