sn4k3/UVtools #558: [Crash] System.InvalidOperationException: Invalid Arrange rectangle.

chriscthornton, 2022-09-20:
I adjusted the resize dimensions under the settings --> resize window tool and I am not able to get inside the settings menu anymore. I downloaded the file again from scratch and it seems that did not work

sn4k3, 2022-09-20:
I can't reproduce this problem. After checking that UVtools work as normal. But there is a bug indeed, that setting is not used anywhere, it should have been lost in any update and so i can't see how it is causing this problem.

What is your OS, resolution and screen scalling? Also can't start UVtools anymore?

chriscthornton, 2022-09-20:
try changing your resolution from 100 and 80 to 1080 and 1080. After I tried this on accident. I wasn't able to use or click on much without uvtools working again.

chriscthornton, 2022-09-20:
I'll paste the crash message. Yes I am on the latest version.

System.InvalidOperationException: Invalid Arrange rectangle.
   at Avalonia.Layout.Layoutable.Arrange(Rect rect) in /_/src/Avalonia.Layout/Layoutable.cs:line 396
   at Avalonia.Layout.Layoutable.ArrangeOverride(Size finalSize) in /_/src/Avalonia.Layout/Layoutable.cs:line 732
   at Avalonia.Controls.WindowBase.ArrangeCore(Rect finalRect) in /_/src/Avalonia.Controls/WindowBase.cs:line 262
   at Avalonia.Layout.Layoutable.Arrange(Rect rect) in /_/src/Avalonia.Layout/Layoutable.cs:line 396
   at Avalonia.Layout.LayoutManager.Arrange(ILayoutable control) in /_/src/Avalonia.Layout/LayoutManager.cs:line 318
   at Avalonia.Layout.LayoutManager.ExecuteInitialLayoutPass() in /_/src/Avalonia.Layout/LayoutManager.cs:line 185
   at Avalonia.Controls.Window.ShowDialog[TResult](Window owner) in /_/src/Avalonia.Controls/Window.cs:line 772
   at Avalonia.Controls.Window.ShowDialog(Window owner) in /_/src/Avalonia.Controls/Window.cs:line 722
   at UVtools.WPF.MainWindow.MenuFileSettingsClicked() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.axaml.cs:line 1088
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Native.PlatformThreadingInterface.SignaledCallback.Signaled(Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 39
   at Avalonia.Native.Interop.Impl.__MicroComIAvnSignaledCallbackVTable.Signaled(Void* this, Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/Interop.Generated.cs:line 4336
--- End of stack trace from previous location ---
   at Avalonia.Native.PlatformThreadingInterface.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 90
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/Program.cs:line 70
Category: Application
File: UVtools_demo_file.sl1  [Version: 0] [Class: SL1File]

# System
UVtools v3.6.5 X64
Operative system: Darwin 21.2.0 Darwin Kernel Version 21.2.0: Sun Nov 28 20:29:10 PST 2021; root:xnu-8019.61.5~1/RELEASE_ARM64_T8101 X64
Processor: 
Processor cores: 8
Memory RAM: 3.07 / 5.62 GB
Runtime: osx.12-x64
Framework: .NET 6.0.9
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Path:       /private/var/folders/m0/rsjh4n255gs8tf_9r9vv2vg00000gn/T/AppTranslocation/F3062E1A-1261-4394-9869-F6D4EC8A25B8/d/UVtools.app/Contents/MacOS/
Executable: /private/var/folders/m0/rsjh4n255gs8tf_9r9vv2vg00000gn/T/AppTranslocation/F3062E1A-1261-4394-9869-F6D4EC8A25B8/d/UVtools.app/Contents/MacOS/UVtools

Machine date time: 09/19/2022 21:45:10
    UTC date time: 09/20/2022 01:45:10

sn4k3, 2022-09-20:
Closing due duplicate of https://github.com/sn4k3/UVtools/issues/515

closed 2022-09-20 by sn4k3