sn4k3/UVtools #566: [BUG] Unable to run due to missing libcvextern

neilenns, 2022-10-02:
## System
 - OS: Win11
 - UVtools version: v3.6.7 X64 [Windows]

## Describe the bug

After upgrading from the prior release I get the following error when running:

Unable to run due one or missing dependencies.
Triggered by: libcvextern (OpenCV)

## Expected behavior

UVTools runs

## Screenshots

[image]

neilenns, 2022-10-02:
I installed by clicking on the "New version v3.6.7 is available!" link in v3.6.6 and just went through the installer.

Uninstalling 3.6.7 and installing 3.6.6 again worked fine.

sn4k3, 2022-10-02:
Yes, it was due a downgrade on library and MSI have this kind of strange problems.
As you said the reinstall is the solution.

scottmudge, 2022-10-03:
On Windows, go to your "Settings > Apps & Features > UVtools > Uninstall", then click the "Repair" button. It will fix it.

For developers, you can modify the MSI manifest/descriptor to always overwrite certain files. I believe the default behavior is to keep the newest file, at least if they share the same GUID. For dependency downgrades, setting it to always overwrite should ensure the downgraded version is installed.

sn4k3, 2022-10-03:
> For developers, you can modify the MSI manifest/descriptor to always overwrite certain files. I believe the default behavior is to keep the newest file, at least if they share the same GUID. For dependency downgrades, this causes problems, so the always-overwrite setting should ensure the downgraded version is installed.

GUID is set to auto (*)
But the stupid thing is that UVtools doesn't upgrade. It uninstall and reinstall. If you notice when you upgrade, UVtools folder, registry and all other components gets removed and installed again. I really don't know but this MSI are a big pain

scottmudge, 2022-10-03:
Ah yeah if it's set to auto it's going to be an issue. I had to manually enter all the files I wanted to unconditionally overwrite, which for a few hundred files was a major pain. 

MSI/WiX is certainly a pain in the --- sometimes.

closed 2022-10-04 by sn4k3