sn4k3/UVtools #567: [BUG]

khoado1, 2022-10-02:
## System
 - OS: macOS Catalina, 2.7 GHZ Quad-Core Intel Core i7
 - Slicer: Lychee Slicer 4.1.0
 - Printer: Elegoo Mars 2 Pro
 - UVtools version: 3.6.7

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

```
UVtools v3.6.7 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jun 21 21:18:39 PDT 2022; root:xnu-6153.141.66~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i7-3740QM CPU @ 2.70GHz
Processor cores: 8
Memory RAM: 8.83 / 16.63 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.9
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 200% (Primary)
    WA: 1440 x 831    UA: 720 x 416

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Mars2Pro-2s-4.5s (0.5x6).ctb [Version: 3] [Class: ChituboxFile]

## Describe the bug
A clear and concise description of what the bug is. --> Unable to Properly Save Profile on Exposure Time Finder

## To Reproduce
Steps to reproduce the behavior:
1. Go to Calibration -> Exposure Time Finder
2. Fill out the profile settings - specifically change the Display width and Display height
3. Click onto the '+' button to save the profile
4. Click onto the Exposure time finder button to generate the test file
5. The action will generate the wrong Display width and Display height but you simply close the application without saving
6. Open the application and go back to the Calibration -> Exposure Time Finder and select the profile that you previously saved
7. Error: You will see that the Display width and height has reverted back to it's previous values.

## Expected behavior

## Screenshots

## Files

 
## Exceptions Log file

## Additional context

sn4k3, 2022-10-02:
If you notice, the display size does not appear on profile text. That means it's not used to restore the profile.
That is by design, because of a simple reason. You can use same profile in different printers without have to revert the display size. 
UVtools auto grab the machine display size from current loaded file and set that to the fields, and then the profile does not make use of the display value. We also asume that the display settings are correct on the file.

PS: Theres is no way to bypass that

khoado1, 2022-10-02:
Understood - ok then the real issue is that the display width and height are reversed.  On an Elegoo Mars 2 Pro the width should be 130.56 mm and the height 82.62 mm.  I only attempted to change the values for that reason.

[image]

sn4k3, 2022-10-03:
Ah ok that a issue with lychee.
I know they do that on some printers, i don't know the reason of flipped XY.
You should contact them and ask.

closed 2022-10-04 by sn4k3