sn4k3/UVtools #568: [BUG] MacOS Crash 3.6.7

ServerOnTheMove, 2022-10-03:
## System
 - OS: MacOS 13.0
 - Slicer: Lychee
 - Printer: 
 - UVtools version: 

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
UVtools v3.6.7 X64
Operative system: Darwin 22.1.0 Darwin Kernel Version 22.1.0: Tue Sep 13 21:11:11 PDT 2022; root:xnu-8792.40.101.181.2~2/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz
Processor cores: 16
Memory RAM: 15.70 / 28.54 GB
Runtime: osx.13-x64
Framework: .NET 6.0.9
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1200 @ 200% (Primary)
    WA: 1920 x 1200    UA: 960 x 600

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: Elder Red Dragon - FotisMint V2_UVT.pwmb [Version: 516] [Class: PhotonWorkshopFile]

## Describe the bug
UVTool Crashes after increasing "Remove Islands" to 50px in "Repair Layer and Issues."

## To Reproduce
Load .pwmb file, click on "Repair", and change "Remove Islands" to 50.
Note - this has worked in an earlier version

## Expected behaviour
[UVtools-2022-10-03-085247.ips.txt](https://github.com/sn4k3/UVtools/files/9697573/UVtools-2022-10-03-085247.ips.txt)

No crash.

## Screenshots
Can't take a screenshot of the crashing app

## Files
 
## Exceptions Log file
Attached

sn4k3, 2022-10-03:
Can't replicate. Can you share the problematic file?

sn4k3, 2022-10-04:
Fixed

closed 2022-10-04 by sn4k3