sn4k3/UVtools #569: [BUG] Repair layers and issues repeatedly crashing on 3.6.7

brummieb, 2022-10-03:
## System
 - OS: Windows 10 Pro 21H2
 - Slicer: Lychee (free)
 - Printer:  Elegoo Mars 3
 - UVtools version:  3.6.7

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

UVtools v3.6.7 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-5930K CPU @ 3.50GHz
Processor cores: 12
Memory RAM: 16.44 / 31.91 GB
Runtime: win10-x64
Framework: .NET 6.0.9
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Screens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2160    UA: 2560 x 1440
2: 3840 x 2160 @ 150%
    WA: 3840 x 2160    UA: 2560 x 1440

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]

```

## Describe the bug
A clear and concise description of what the bug is.

I've recently upgraded to v3.6.7, and now every time I use "repair layers and issues" to fix just islands the process runs for a few seconds then UVTools exits (no error messages - the window just disappears). It was working in the previous version I was running (can't remember the exact version number - sorry)

## To Reproduce
Steps to reproduce the behavior:

1. Tools->Repair layers and issues
2. Untick "Repair resin traps" (leaving "Repair islands" and "Remove empty layers" ticked)
3. All other options as default (5 px islands, 4 layers recursive, 2 layers attach)
4. Start the repair
5. Process will run for a few seconds then UVTools windows disappears (assumed crashed). The progress dialog is showing "Removed recursive islands" when UVTools crashes.

Ticking "Remove empty layers" only seems to be fine, as is "Repair resin traps" only.

## Expected behavior

Not crash ;)

## Files
 
I tried to reproduce the bug with a simple sliced cube, but it did not crash; so it does appear to be file dependant. The smallest of the "bad" files is >60MB, but I can provide it if necessary.

## Exceptions Log file

No AppData\Local\UVtools\errors.log present unfortunately

sn4k3, 2022-10-03:
Maybe related to https://github.com/sn4k3/UVtools/issues/568 ?
I also can not reproduce, can you submit a file where i can?

brummieb, 2022-10-03:
Could be related. I just tried again and confirmed that the dialog runs through showing "Removed recursive islands", and after a few seconds it pauses (still showing "Removed recursive islands") then crashes.

The file is too big to attach (63MB) so I've uploaded it here: https://spikyfish.com/test.zip

It's not a "pretty" slice; I'm a resin novice, and I've crammed a lot onto the build plate, with overlapping rafts (though I've done the same before and UVTools was fine with it).

PS I manually went through the file above in UVTools, removing small islands (no crashes). I notice that it appears to incorrectly pick up a couple of areas as islands; even though there were supports in the layers below. I used the paint tool to then add more supports but they were still detected as islands.

sn4k3, 2022-10-03:
Ok i found and fix the problem.

>  I notice that it appears to incorrectly pick up a couple of areas as islands; even though there were supports in the layers below.

This may be caused by unsafe supports, not to meet criteria. Can you indicate layer number and screenshot of position?

closed 2022-10-04 by sn4k3

brummieb, 2022-10-04:
Confirmed fixed - many thanks.

> This may be caused by unsafe supports, not to meet criteria. Can you indicate layer number and screenshot of position?

That does sound likely yes. If I run an initial "Repair layers and issues" (with default settings) I can see (amongst other islands) two at layers 100 and 110. The island on layer 100 only has a single support underneath, but the one on layer 110 has two.

This is the 99-100 difference:

[image]

And the 109-110 difference:

[image]

sn4k3, 2022-10-04:
Ok it's a threshold "problem", supports by default must be at least 0.25x the island mass (Configurable) on settings

brummieb, 2022-10-04:
Great - thanks. No bug then.

Off topic (and admittedly probably better as a separate ticket); UVTools shows me that the automatic support generation in slicers misses loads of obvious islands; presumably because UVTools is checking the layers at the pixel level. Would it be possible for UVTools to do the job of supporting models? I.e. position your models on the build plate using the slicer software, then export without adding supports, and let UVTools do the job?