sn4k3/UVtools #571: [BUG] Huge problem for release 3.6.8. It doesn't start....

tecchese, 2022-10-04:
Release 3.6.8 does not start...simply...
See pics, previous version (3.6.7) works yet...

[image]
[image]
[image]

sn4k3, 2022-10-04:
May be related with https://github.com/sn4k3/UVtools/issues/566

Try to uninstall and install, or repair by rerun installer

scheini79, 2022-10-05:
> May be related with #566
> 
> Try to uninstall and install, or repair by rerun installer

That fixed it for me, just rerun the installer and choosed repair.

tecchese, 2022-10-05:
yes, it is the unique solution...uninstalled, rebooted e installed...

sn4k3, 2022-10-12:
Fixed v3.7.0

closed 2022-10-12 by sn4k3