sn4k3/UVtools #573: Detecting islands that aren't islands

SgtG294, 2022-10-05:
After the latest updates, my UVTools has been detecting islands that are actually connected to one or more supports

sn4k3, 2022-10-05:
Link a sample file, indicate layer number and island position with a screenshot

SgtG294, 2022-10-05:
[image: image.png]

Layer 60 is the first group of the issue but all major islands are
supported throughout.
 88mm gun, crew PRESUPPORTED DC.pwmb
<https://drive.google.com/file/d/1QQGz9WEjnQAPVIdzplSJGSt_M9caOzkH/view?usp=drive_web>

sn4k3, 2022-10-05:
Uploads and attachments via email don't work here. You need to use github directly to post content.

dprossner, 2022-10-11:
I'm having the same issue on 3.6.8 and 3.6.7. attached is a file sliced in lychee as .ctb and converted to zip in uvtools (GitHub doesn't support .ctb). I have already run the automated island repair tool but it is still falsely flagging areas on layer 100 as islands even though they are supported on the previous layer. anti-aliasing settings in the slicer don't seem to make a difference and this behavior wasn't present on 3.6.6 also included screenshots of the same file on the same layer on 3.6.6 and 3.6.7
![3_6_6](https://user-images.githubuser
[islander.zip](https://github.com/sn4k3/UVtools/files/9751532/islander.zip)
content.com/65640171/195007961-09c349ed-d12b-4377-84a0-a3d3a6fcb565.png)
[image]

rw-donaldp, 2022-10-11:
Just to confirm that I'm having similar issues with false island detection with 3.6.7 and 3.6.8. I've reverted to 3.6.6 as the later releases are not workable for my current project.

closed 2022-10-19 by sn4k3

antwal, 2022-10-30:
> I see no problem.
> 
> [image]
> 
> Insider: [image]
> 
> First island have a mass of **145px**. Supports contact points below have a total mass of **13px** Giving default safe threshold (0.25x) gives you: **145 * 0.25 = 36px** Due of you have **13px** supporting a mass that requires at least **36px** it detect as island, you need at least 23px more in the support area to it not trigger. In other words, your supports aren't enough/safe giving the threshold settings. So nothing wrong with new UVtools, problem was on the older versions.
> 
> Older UVtools had various bugs that cause this detection to pass
> 
> 1. More area detected as it use x*y bouding box
> 2. A overhang bad check that cause to discard the island.
> 
> So new version fixes the problem. You can adjust threshold if you want that to pass. eg: 0.08x

so we have to manually set the size and change it every time for each type of print?

the thing is very uncomfortable if that were the case

antwal, 2022-10-30:
I was able to compare versions 3.6.3 and version 3.8.0 after having configured it correctly; there are small differences but on the latest version I find non-existent problems such as islands and overhangs

I am attaching the screen to the same layer

[image]
[image]

sn4k3, 2022-10-30:
Overhang detection was improved with a cross kernel which fixes the bug where your setting was resulting in more pixels than selected, ignoring possible overhangs. If you want you can increase the px threshold of the overhangs.

Islands discard logic with overhangs was brought back BUT i set a threshold, which is, need at least 1/4 of supporting pixels. That is your setting (0.25 / 4). This helps on prevent possible islands to be ignore.
`pixelsSupportingIsland >= requiredSupportingPixels / 4` if not it's considered an insland. 

All i can do is allow user to define the divisor, so they can opt for more or less error. Keep in mind the detections are not perfect, and sometimes is better to have a well-defined threshold and a agressive discarding threshold and lose possible islands. Those who are not issues you can ignore them.