sn4k3/UVtools #577: [BUG] error on loading cws files.

rflorean, 2022-10-09:
## System
 - Windows 10 x64 v20H2
 - Slicer: novamaker v.2.6.6 64bit
 - Printer: nova3d bene4
 - UVtools version: 3.6.4

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

System.IO.FileLoadException: Layer 0 not found
   at UVtools.Core.FileFormats.FileFormat.DecodeLayersFromZipRegex(ZipArchive zipArchive, String regex, IndexStartNumber layerIndexStartNumber, OperationProgress progress, Func`3 matGenFunc) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 3645
   at UVtools.Core.FileFormats.CWSFile.DecodeInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\CWSFile.cs:line 841
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 3416
   at UVtools.WPF.MainWindow.<>c__DisplayClass107_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1367

## Describe the bug
A clear and concise description of what the bug is.
On open file cws the error comes out.

## To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior
<!--
A clear and concise description of what you expected to happen.
!-->

## Screenshots
<!--
If applicable, add screenshots to help explain your problem. Never hide information! Full GUI/window screenshot is always a must!
If bug is related to GUI/visuals print a screenshot from Help -> About
!-->

## Files
<!--
Were possible attach the problematic file, both source and output file if applicable.
!-->
 
## Exceptions Log file
<!--
In some unexpected exceptions errors and application crash it will generate and save a log file under AppData\Local\UVtools\errors.log. 
To navigate to the error.log folder you can open UVtools, go to Help -> Open settings folder
If present you can attach the log file or the content here.
!-->

## Additional context
<!--
Add any other context about the problem here.
!-->

<!--
# Overview:
1. Use descriptive and complete title! 
2. Review all information and make sure it's clear and complete.
3. Erase all unused sections.
!-->

sn4k3, 2022-10-09:
Layer 0 not found on the file.
Include the file that caused that so i can look

sn4k3, 2022-10-09:
The problem looks like with filename... If you are using a number on the file end name that raise problem with UVtools. Do not use file with numbers in the end

rflorean, 2022-10-10:
Thanks for your reply. I send you what happens:
first test cws sliced with novamaker
second test cws sliced with lycheeslicer

Regards

sn4k3, 2022-10-10:
Attachments via email don't work here. You have to use github directly on web browser

rflorean, 2022-10-10:
https://user-images.githubusercontent.com/42885977/194930830-39deb8c5-067a-4033-9b1c-8cdc5dfd87f7.mp4

https://user-images.githubusercontent.com/42885977/194930841-db1e0eb5-eb2d-4d3a-9dd5-18cd091482e2.mp4

cws created with novamaker doesn't work; cws created with lycheeslice does work! same content

rflorean, 2022-10-10:
original stl file
[_221010_ganasce_mobili_H110_4_centrali_2_2_BCDEF_55_10_gradi_passo_9.zip](https://github.com/sn4k3/UVtools/files/9748228/_221010_ganasce_mobili_H110_4_centrali_2_2_BCDEF_55_10_gradi_passo_9.zip)

sn4k3, 2022-10-10:
I need the sliced files (.cws) to explore the contents.
I confirm that .cws from original slicer does have problem with UVtools which will be fixed on next release, but the error I see from your first video looks like the problem with filename. Either way both will be fixed on next release.

rflorean, 2022-10-11:
this was produced with novamaker: [test_uvtools.zip](https://github.com/sn4k3/UVtools/files/9751514/test_uvtools.zip)
the following file has been produced with lycheeslicer
[second_test_with_LycheeSlicer.zip](https://github.com/sn4k3/UVtools/files/9751523/second_test_with_LycheeSlicer.zip)

rflorean, 2022-10-11:
both have same origin, so they should be identical but it is not true.

sn4k3, 2022-10-11:
Here the problem, as told it's a file name problem, it have a number in the end: "_221009_ganasce_mobili_H110_4_centrali_2_2_BCDEF_55_10_gradi_passo_9" that cause to parse layers incorrectly.

[image]

rflorean, 2022-10-11:
Crazy! Stl file name has a number at the end! In my life I've written a lot of programs, but this looks me crazy: but it is true. I've changed stl file name, created with novamaker new cws, and then, oops, uvtools reads the cws file. This issue doesn't happen with lycheeslice. Thanks. Regards

sn4k3, 2022-10-11:
>  This issue doesn't happen with lycheeslice.

Because lychee uses no filename as layer image, it uses lychee000.png
[image]

rflorean, 2022-10-11:
👍

Ottenere Outlook per Android<https://aka.ms/AAb9ysg>

[image]<https://user-images.githubusercontent.com/113281/195090799-abc395e9-5b21-4045-b2be-892df58cc139.png>

sn4k3, 2022-10-12:
Fixed

closed 2022-10-12 by sn4k3