sn4k3/UVtools #579: [BUG] "Edit print parameters" window too large for 1080p screen on Win10

Maeyanie, 2022-10-12:
## System
 - OS: Windows 10 x64 Build 19044
 - Slicer: PrusaSlicer
 - Printer: Phrozen Mini 8K
 - UVtools version: 3.7.0

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:

UVtools v3.7.0 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
Processor cores: 12
Memory RAM: 21.46 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       D:\Program Files\UVtools\
Executable: D:\Program Files\UVtools\UVtools.exe
Loaded file: Quetzalcoatlus Mounted.encrypted.ctb [Version: 0] [Class: CTBEncryptedFile]

```

## Describe the bug
On my 1080p screen, the "Edit print parameters" screen is too large and the buttons are hidden by the task bar.

## To Reproduce
Steps to reproduce the behavior:
1. Tools
2. Edit print parameters

## Expected behavior
[image]

## Screenshots
[image]

sn4k3, 2022-10-12:
That's strange. On my laptop (720p) scales correctly:

[image]

Do you have any window manager program installed?

Maeyanie, 2022-10-12:
Yep, you're right, that did work. Bit of an odd interaction, but I guess it's okay. Thanks.

closed 2022-10-12 by Maeyanie

sn4k3, 2022-10-12:
> Yep, you're right, that did work. Bit of an odd interaction, but I guess it's okay. Thanks.

There's a reason. If resize is allowed, windows can't have a MaxWidth / MaxHeight, or else you can't resize more than that size and the option is useless, even maximize wont pass the MaxWdith/Height, and so it must be "infinite".

With that option off it will constrain windows to screen size and resize.