sn4k3/UVtools #580: [BUG] -

draj1234, 2022-10-12:
## System
 - OS: Windows X64 - 22H2(OS Build 22621.674
 - Slicer:  Not gone yet there
 - Printer: Not gone yet there 
 - UVtools version: 3.7.0

```
1. Go to "UVtools -> Help -> About"
2. Right click on "Copy information" -> "Copy the essential information"
3. Paste that infomation bellow this line:
---------------------------------------------------
UVtools v3.7.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx  
Processor cores: 8
Memory RAM: 6.17 / 9.94 GB
Runtime: win10-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]

```

## Describe the bug
The Fantastic Feature of " PCB Exposure" in UVtools - throws error as below when we load a gerber file.
I am telling you thsi feature is a killer feature for Electronics enthusiast - Pleas ehelp to fix these issues , so it is very usefull for the community.

## To Reproduce
Steps to reproduce the behavior:
1. Go to 'PCB Exposure in Tools menu'
2. Click on 'Add the Gerber file' and add the attached gerber file
3. Press the PCB exposure (all other settings on that page is left as default)
4. While converting it throws error consistently as below
----------------------------------------------------------
Error Thrown:
=========
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection. (Parameter 'index')
   at System.Collections.Generic.List`1.get_Item(Int32 index)
   at UVtools.Core.Gerber.Primitives.OutlinePrimitive.DrawFlashD3(Mat mat, PointF at, MCvScalar color, LineType lineType) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Gerber\Primitives\OutlinePrimitive.cs:line 90
   at UVtools.Core.Gerber.Apertures.MacroAperture.DrawFlashD3(Mat mat, PointF at, MCvScalar color, LineType lineType) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Gerber\Apertures\MacroAperture.cs:line 34
   at UVtools.Core.Gerber.GerberDocument.ParseAndDraw(String filePath, Mat mat, SizeF xyPpmm, GerberMidpointRounding sizeMidpointRounding, Boolean enableAntialiasing) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Gerber\GerberDocument.cs:line 416
   at UVtools.Core.Operations.OperationPCBExposure.GetMat(String file) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationPCBExposure.cs:line 259
   at UVtools.Core.Operations.OperationPCBExposure.ExecuteInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\OperationPCBExposure.cs:line 267
   at UVtools.Core.Operations.Operation.Execute(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Operations\Operation.cs:line 732
   at UVtools.WPF.MainWindow.<>c__DisplayClass119_0.<RunOperation>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 2162
-------------------------------------------------------------------------------------------------------------------------

## Expected behavior

## Files

draj1234, 2022-10-12:
This Gerber was generated using Kicad - Application: KiCad (64-bit)
Version: (6.0.7), release build
Libraries:
	wxWidgets 3.1.7
	libcurl/7.83.1-DEV Schannel zlib/1.2.12
Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
	Date: Jul 26 2022 02:49:38
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI
Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

draj1234, 2022-10-12:
[OB2.0_BT_MODULE-F_Cu.zip](https://github.com/sn4k3/UVtools/files/9765935/OB2.0_BT_MODULE-F_Cu.zip)

sn4k3, 2022-10-12:
Found the problem. Will be fixed on next patch.

[image]

sn4k3, 2022-10-17:
Fixed

closed 2022-10-17 by sn4k3

draj1234, 2022-10-17:
Super ! thanks - now I tested the same gerber file (which I had posted earlier which refused to open) and now with this new version V3.7.1 of UVtools it is converting it correctly ! thanks once again for fixing the same.

74HC14, 2022-10-26:
Just in case anyone else has this problem, I'll note that I was having difficulties getting my gerbers to show up. They'd just be fully black, zero pixels of white. The solution was to move them up close to coordinates 0,0, apparently they were out of the bounds of UVtools.

sn4k3, 2022-10-26:
> Just in case anyone else has this problem, I'll note that I was having difficulties getting my gerbers to show up in UVtools. They'd just be fully black, zero pixels of white. The solution was to move them up close to coordinates 0,0 in KiCAD, apparently they were out of the bounds of UVtools.

UVtools place Gerber on it exact drawn coordinates. If is out of LCD range it will not show.
It can crop the board but you need to include the outline gerber and still it will draw on original coordinates first before crop and move