sn4k3/UVtools #582: x[Crash] Emgu.CV.Util.CvException: OpenCV: s >= 0

i11, 2022-10-14:
Emgu.CV.Util.CvException: OpenCV: s >= 0
   at Emgu.CV.CvInvoke.CvErrorHandler(Int32 status, IntPtr funcName, IntPtr errMsg, IntPtr fileName, Int32 line, IntPtr userData)
Category: Non-UI
File: 12mm-heavy1.ctb  [Version: 4] [Class: ChituboxFile]

# System
UVtools v3.7.0 Arm64
Operative system: Darwin 21.6.0 Darwin Kernel Version 21.6.0: Mon Aug 22 20:20:05 PDT 2022; root:xnu-8020.140.49~2/RELEASE_ARM64_T8101 Arm64
Processor: Apple M1
Processor cores: 8
Memory RAM: 6.43 / 11.91 GB
Runtime: osx.12-arm64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools

Machine date time: 10/15/2022 01:14:53
    UTC date time: 10/14/2022 23:14:53

sn4k3, 2022-10-14:
I have no clue from where it comes.
What action cause this crash?
Had you free RAM at that time?

i11, 2022-10-15:
Seems like a specific ctb file related issue. It doesn't crash with others. I can upload it somewhere if you would like to check it out.
I don't think I had any RAM issues.
Some pre-history to this:
UVTools stopped working on me after another brew update, it would always crash with missing OpenCV lib. I came across `ffmpeg@4` dependencies and did the downgrade. After that UVTools would load, and it just happened to crash on this first ctb file.

sn4k3, 2022-10-15:
brew may install the arm version which apple locks without sign the app, i don't know how you are running that. You can retry with x64 package here from github and see if error still happen.

If you want to send the file you can use any file upload service and paste here the link, or zip it and upload here if not too big.

PS: Downgrade ffmpeg@4 is requied for arm only

closed 2022-11-09 by sn4k3