sn4k3/UVtools #587: [FEATURE REQUEST] .PM3R files support

hkjytkd, 2022-10-19:
Support for .PM3R files (Anycubic Photon M3 Premium new files)

Needed support for PM3R files, the new format for the Anycubic Photon M3 Premium printers just released from Anycubic.
CHITUBOX & TANGO has already the support to these new printers but when sending the files to UV Tools nothing happens.

sn4k3, 2022-10-19:
Chitubox v1.9.4 don't have it
Anycubic photon workshop v2.2.16 don't have it
Google search gives very few results (almost to none)
Anycubic.com website don't have it
I can only find it on latest Tango

Is the printer available to the public at the moment?

hkjytkd, 2022-10-20:
Sorry, chitubox does not support it, but tango v 2.11.99.48 and anycubic photo workshop v2.2.19 support it，the machine is now on sale。

hkjytkd, 2022-10-20:
I bought the installation package that came with the device.

How can I send it to you?

sn4k3, 2022-10-20:
Upload to a web service, eg https://wetransfer.com and post the link here

hkjytkd, 2022-10-20:
https://we.tl/t-ea68hlmEFn

hkjytkd, 2022-10-20:
Please try to download!

sn4k3, 2022-10-20:
It works, will be in the next release

sn4k3, 2022-10-20:
Implemented

closed 2022-10-20 by sn4k3