sn4k3/UVtools #590: [Crash] System.Runtime.InteropServices.COMException (0x80004005): No such interface supported

AntonyGibsonFoster, 2022-10-24:
System.Runtime.InteropServices.COMException (0x80004005): No such interface supported
 ---> System.Runtime.InteropServices.COMException (0x80004002): GetItemAt failed
   at Avalonia.Win32.Win32Com.Impl.__MicroComIShellItemArrayProxy.GetItemAt(Int32 dwIndex) in /_/src/Windows/Avalonia.Win32/Win32Com/Win32.Generated.cs:line 439
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass2_0.<ShowFileDialogAsync>b__0() in /_/src/Windows/Avalonia.Win32/SystemDialogImpl.cs:line 102
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass2_0.<ShowFileDialogAsync>b__0() in /_/src/Windows/Avalonia.Win32/SystemDialogImpl.cs:line 121
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at UVtools.WPF.MainWindow.OpenFile(Boolean newWindow, FileDecodeType fileDecodeType) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1015
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application
File: UVtools_demo_file.sl1  [Version: 0] [Class: SL1File]

# System
UVtools v3.7.2 X64
Operative system: Microsoft Windows 10.0.18363 X64
Processor: Intel(R) Xeon(R) CPU E3-1240 v3 @ 3.40GHz
Processor cores: 8
Memory RAM: 7.82 / 15.94 GB
Runtime: win10-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Path:       C:\Program Files (x86)\UVtools\
Executable: C:\Program Files (x86)\UVtools\UVtools.exe

Machine date time: 10/24/2022 11:36:50
    UTC date time: 10/24/2022 10:36:50

AntonyGibsonFoster, 2022-10-24:
Error occurs when I try to open a Chitu sliced file for the Anycubic M3 Max, ie .p3m3:
UVTools 3.7.1 was installed and working fine on Friday. I came to use it this morning and ran into this exception, so I uninstalled 3.7.1 and installed the latest 3.7.2 but I'm still receiving this error.

I've also tried to build the latest version with visual studio 2022 and test from there, but I received the same unexpected application error.

AntonyGibsonFoster, 2022-10-24:
So I had been trying for a while to be able to open a p3m3 file with UVTools 3.7.1 & 2 and receiving the above error. The proceedure was: Open UVTools, file\open\'somefile'.p3m3 and then crash. After getting frustrated, and just to try something else, I tried opening a .zip file. This worked perfectly and following on from that I am able to once again open .p3m3 files again without issue. I don't know if this is useful to debugging, but it seemed that opening the .zip reset something in the UVTools, and now everything works again.

sn4k3, 2022-10-24:
This is related to UI Framework and out of UVtools scope.
Some users may had trouble when loading files from pen drive or libraries. Try to open files from absolute path instead using the explorer libraries.

AntonyGibsonFoster, 2022-10-24:
The file I was opening was on an external HDD and was opened from it's absolute path in UVTools. Like I said though, the problem has now gone after opening a sliced .zip file; It was like something just reset somewhere and now eveything is working properly again.

sn4k3, 2022-10-24:
It's wierd bugs from framework and out of my scope, lets hope when I upgrade the 0.11.x stable version that kind of problems go away.

AntonyGibsonFoster, 2022-10-24:
UVTools is fantastic and I'm happy to use this workaround if it happens again. Thank you for all your hard work and your very quick reply to this issue.

sn4k3, 2023-09-17:
Maybe fixed in last version, reopen if needed

closed 2023-09-17 by sn4k3