sn4k3/UVtools #591: [BUG] 3.7.2 nonexistent islands

antwal, 2022-10-27:
## System
 - OS: macOS Catalina 10.15.7
 - Slicer: Lychee Slicer
 - Printer: Elegoo Saturn S
 - UVtools version: 3.7.2

```
UVtools v3.7.2 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jun 21 21:18:39 PDT 2022; root:xnu-6153.141.66~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9900K CPU @ 3.60GHz
Processor cores: 16
Memory RAM: 22.59 / 62.96 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.5.4-235

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 200% (Primary)
    WA: 2560 x 1417    UA: 1280 x 708
2: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708
3: 2560 x 1440 @ 200%
    WA: 2560 x 1417    UA: 1280 x 708

Path:       /Applications/UVtools 3.7.2.app/Contents/MacOS/
Executable: /Applications/UVtools 3.7.2.app/Contents/MacOS/UVtools
Loaded file: Ashe_Others_2.ctb [Version: 2] [Class: ChituboxFile]

```

## Describe the bug
nonexistent islands are found, two examples attached in the screenshots

## Screenshots
[image]
[image]

[image]
[image]

antwal, 2022-10-27:
with the previous version I used 3.6.3 it worked perfectly

[image]

I also tried 3.7.1 but it generates the same problem, it signals islands that don't really exist

[image]

antwal, 2022-10-27:
I have also tried with other versions but the same problem occurs

3.7.0 same problem
[image]

3.6.8 same problem
[image]

antwal, 2022-10-27:
To verify the problem I am attaching the ctb file: [Download](https://file.io/ll07FaxG4PAF)

sn4k3, 2022-10-27:
Duplicate of https://github.com/sn4k3/UVtools/issues/573 -- Read explaination

closed 2022-10-30 by sn4k3