sn4k3/UVtools #598: [BUG] System.IndexOutOfRangeException when processing a file

neilenns, 2022-11-04:
## System
 - OS: Win11
 - Slicer: Chitubox v1.9.4
 - Printer: Mars 2 Pro Mono
 - UVtools version: 3.8.1

```
UVtools v3.8.1 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 5 5600X 6-Core Processor             
Processor cores: 12
Memory RAM: 11.77 / 31.91 GB
Runtime: win10-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 150% (Primary) (On this)
    WA: 3840 x 2088    UA: 2560 x 1392
2: 1920 x 1080 @ 100%
    WA: 1920 x 1032    UA: 1920 x 1032
3: 1280 x 1024 @ 100%
    WA: 1280 x 976    UA: 1280 x 976

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: _Square_blank_cap.ctb [Version: 4] [Class: ChituboxFile]

```

## Describe the bug
System.IndexOutOfRangeException when processing a file.

```
System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at UVtools.Core.Managers.IssueManager.DetectIssues(IslandDetectionConfiguration islandConfig, OverhangDetectionConfiguration overhangConfig, ResinTrapDetectionConfiguration resinTrapConfig, TouchingBoundDetectionConfiguration touchBoundConfig, PrintHeightDetectionConfiguration printHeightConfig, Boolean emptyLayersConfig, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\Managers\IssueManager.cs:line 0
   at UVtools.WPF.MainWindow.<>c__DisplayClass214_0.<ComputeIssues>b__0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.Issues.cs:line 660
```

File is attached.
[_Square_blank_cap.zip](https://github.com/sn4k3/UVtools/files/9941021/_Square_blank_cap.zip)

## Expected behavior
No exception

sn4k3, 2022-11-04:
You have a ton of empty layers on middle of the file. Is that on purpose?

[image]

neilenns, 2022-11-04:
It's a file issue where I forgot to generate the supports so the models are just floating in air above the rafts. Wouldn't expect to see an exception in the app though.

neilenns, 2022-11-04:
Ha, I was just typing as you were :)

sn4k3, 2022-11-04:
I was unable to reproduce the bug, it remove empty layers correctly without crash.

My workflow:

- Open file
- Detect all issues
- Run auto repair

neilenns, 2022-11-04:
It happens every time I open it using v3.8.1 🤷‍♂️

At any rate, I've added the supports in :)

sn4k3, 2022-11-04:
Then you should be using a different configuration than I.
Attach you setting file here by going to Help -> Open settings folder

neilenns, 2022-11-04:
[suggestions.zip](https://github.com/sn4k3/UVtools/files/9941069/suggestions.zip)

sn4k3, 2022-11-04:
Thats the `suggestions`, i need the `usersettings`

neilenns, 2022-11-04:
> Thats the `suggestions`, i need the `usersettings`

There is no `usersettings` in the folder. Just `recentfiles.dat` and `suggestions.xml`.

sn4k3, 2022-11-04:
Ok that mean you are with stock settings, let me try that...

sn4k3, 2022-11-04:
Found the problem.

Is the combination of compute issues when the tab is open for the first time and your suggestions auto apply.
To prevent this disable this last box:

[image]

Will fix on next patch

neilenns, 2022-11-04:
Glad you were able to reproduce it!

sn4k3, 2022-11-05:
Fixed

closed 2022-11-05 by sn4k3

neilenns, 2022-11-10:
Confirmed this no longer happens in 3.8.3. Yes, I accidentally generated a file without supports again 🤦