sn4k3/UVtools #600: Pads that are non-orthogonal do not import in PCB Exposure

grdrgrd, 2022-11-04:
System

    OS: Windows 10 x64 v21H2
    Slicer: N/A
    Printer: Phrozen Sonic Mega 8K
    UVtools version: 3.8.1

UVtools v3.8.1 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
Processor cores: 8
Memory RAM: 7.08 / 15.90 GB
Runtime: win10-x64
Framework: .NET 6.0.10
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 125% (Primary) (On this)
    WA: 1920 x 1042    UA: 1536 x 834

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: Mega8K_ResTestREV3_60s_Neg_RoundUp.ctb [Version: 0] [Class: CTBEncryptedFile]

Describe the bug

I tried to import a Gerber file using the PCB Exposure tool (UVTools version 3.8.1) and found that the pads that are not orthogonal were not imported. From inspecting the Gerber file, it appears that the angled pads are defined using macro apertures, which may be the cause of the issue.

To Reproduce

Steps to reproduce the behavior:

    Open UVTools (GUI)
    Open a sliced file previously created for my printer (.ctb in my case)
    Select Tools -> PCB exposure
    Add Gerber file (attached), leave all settings default, then generate the traces
    See the lack of apertures in the imported image

Expected behavior

All 8 pads should be imported as shown below.
Screenshots

This is how I would expect the file to render:

[image]

Files

This is the Gerber file I am trying to import.
[PCB1.zip](https://github.com/sn4k3/UVtools/files/9942323/PCB1.zip)

sn4k3, 2022-11-04:
Found the problem, parsing bug

[image]

All `%` alone on a new line is causing the trouble

grdrgrd, 2022-11-05:
Good to know. I edited my Gerber file so that the `%` is moved to the end of the previous line. Importing the edited file to UVTools worked perfectly.

sn4k3, 2022-11-05:
Fixed

closed 2022-11-05 by sn4k3