sn4k3/UVtools #602: [FEATURE REQUEST] Difference view with multiple layers.

bartoszek, 2022-11-08:
## Is your feature request related to a problem? Please describe
Would be nice to have ability to view multiple layers in the difference view to assess how many layers there's till island gets reconnected without having to slowly cycle through a bunch of layers.

## Describe the solution you'd like
Add a two alternating colors with a value gradient, showing multiple layers in the difference view.

## Screenshots
Example of 4 layers difference view.
[image]

bartoszek, 2022-11-09:
Nex step would be to add voxel rendering to inspect print file in glorious 3d 😏

bartoszek, 2022-11-11:
> the 3d mesh reconstruction from 2D is very slow and generate ton of triangles

No no no, just do a voxel rendering, like the VTK does.

sn4k3, 2022-11-11:
openCV compiles the VTK module but never tried, documentation is lacking.
An extra problem is to make the render to work integrated, it must be displayed inside current UI framework and integrate with other elements which is no easy task. I'm still waiting for thier openTK implementation, currently openGL can't be used on windows bug-free

sn4k3, 2026-09-19:
Implemented in v7

closed 2026-09-19 by sn4k3