sn4k3/UVtools #603: [BUG] 3.8.x builds do not run under Debian bullseye

wohali, 2022-11-08:
## System
 - OS: Debian "bullseye" 11.5
 - Slicer: n/a
 - Printer: n/a
 - UVtools version: 3.8.x (3.8.0+ including 3.8.2)

## Describe the bug

With the update to openCV 4.6.0 (unavailable on `bullseye` or `bullseye-backports`) the AppImage fails due to ABI errors, as `bullseye`'s `libc` is 2.31:

```
Your system doesn't have the required dependencies in order to run.
Those dependencies are required at libcvextern/OpenCV library.
UVtools is built on top of the OpenCV and therefore cannot run.

Missing dependencies:
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)
/tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so: /lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by /tmp/.mount_UVtool6QzGsT/usr/bin/libcvextern.so)

Please install or build the dependencies in order to run the software.
Check the manual page at 'Requirements' section
```

This may also affect Ubuntu 22.04 (LTS) - it has libc 2.35, not in the AppImage list above.

## To Reproduce
Steps to reproduce the behaviour:
1. Run debian bullseye
2. Attempt to run the AppImage or install using the `install-uvtools.sh` script.

## Expected behavior
Program should run.

## Screenshots
[image]

## Workaround
v3.7.2 continues to work just fine.

sn4k3, 2022-11-08:
Ubuntu 22.04 works fine as it use latest libc
Newest opencv use newer libc and then it will fail on outdated systems.

Solution: You need to update your libc OR compile libcvextern on your system to link against your libc version

wohali, 2022-11-08:
No luck, I'm afraid:

```
# apt-cache policy libc6
libc6:
  Installed: 2.31-13+deb11u5
  Candidate: 2.31-13+deb11u5
  Version table:
 *** 2.31-13+deb11u5 500
        500 http://debian.mirror.iweb.ca/debian bullseye-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     2.31-13+deb11u4 500
        500 http://debian.mirror.iweb.ca/debian bullseye/main amd64 Packages
```

I will attempt a full source compile and report back.

sn4k3, 2022-11-08:
btw have you tried: `apt full-upgrade` ?

wohali, 2022-11-08:
yes, debian bullseye (stable) doesn't have any newer libc6, not even in backports:

https://packages.debian.org/search?keywords=libc6&searchon=names&suite=stable&section=all

[image]

wohali, 2022-11-08:
opencv.org is down right now so I can't test a full compile of 4.6.0 today. I will come back to this in the future. Thanks for your considerations :)

sn4k3, 2022-11-08:
You can't use opencv repo, use my [script](https://github.com/sn4k3/UVtools/blob/master/build/libcvextern.sh)

run as: `bash libcvextern.sh 4.6.0`

wohali, 2022-11-08:
Looks successful, until it gets to:

```
MSBuild version 17.3.2+561848881 for .NET
/usr/share/dotnet/sdk/6.0.403/Sdks/NuGet.Build.Tasks.Pack/build/NuGet.Build.Tasks.Pack.targets(221,5): error : Could not find a part of the path '/home/joant/temp/UVtools/build/emgucv-4.6.0/libs/runtimes/linux-arm64/native'. [/home/joant/temp/UVtools/build/emgucv-4.6.0/Emgu.CV.Platform/NetStandard/Emgu.CV.Platform.NetStandard.csproj]
make[2]: *** [platforms/nuget/Emgu.CV.runtime.debian-arm64/CMakeFiles/Emgu.CV.runtime.debian-arm64.nuget.dir/build.make:78: platforms/nuget/Emgu.CV.runtime.debian-arm64/CMakeFiles/Emgu.CV.runtime.debian-arm64.nuget] Error 1
make[1]: *** [CMakeFiles/Makefile2:3736: platforms/nuget/Emgu.CV.runtime.debian-arm64/CMakeFiles/Emgu.CV.runtime.debian-arm64.nuget.dir/all] Error 2
make: *** [Makefile:182: all] Error 2
```

There is a `emgucv-4.6.0/libs/libcvextern.so`, and all of its external dependencies are valid:

```
$ ldd emgucv-4.6.0/libs/libcvextern.so 
	linux-vdso.so.1 (0x00007ffd6ce43000)
	libgeotiff.so.5 => /lib/x86_64-linux-gnu/libgeotiff.so.5 (0x00007f5b69463000)
	libdc1394.so.25 => /lib/x86_64-linux-gnu/libdc1394.so.25 (0x00007f5b693ea000)
	libavcodec.so.58 => /lib/x86_64-linux-gnu/libavcodec.so.58 (0x00007f5b67e32000)
	libavformat.so.58 => /lib/x86_64-linux-gnu/libavformat.so.58 (0x00007f5b67bad000)
	libavutil.so.56 => /lib/x86_64-linux-gnu/libavutil.so.56 (0x00007f5b678ec000)
	libswscale.so.5 => /lib/x86_64-linux-gnu/libswscale.so.5 (0x00007f5b67851000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5b6784b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5b67829000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5b6765c000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5b67518000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5b674fe000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5b67327000)
	libtiff.so.5 => /lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f5b672a1000)
	libproj.so.19 => /lib/x86_64-linux-gnu/libproj.so.19 (0x00007f5b66f2e000)
	libraw1394.so.11 => /lib/x86_64-linux-gnu/libraw1394.so.11 (0x00007f5b66f1d000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f5b66eff000)
	libswresample.so.3 => /lib/x86_64-linux-gnu/libswresample.so.3 (0x00007f5b66edd000)
	libvpx.so.6 => /lib/x86_64-linux-gnu/libvpx.so.6 (0x00007f5b66cc0000)
	libwebpmux.so.3 => /lib/x86_64-linux-gnu/libwebpmux.so.3 (0x00007f5b66cb4000)
	libwebp.so.6 => /lib/x86_64-linux-gnu/libwebp.so.6 (0x00007f5b66c4b000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f5b66c23000)
	libdav1d.so.4 => /lib/x86_64-linux-gnu/libdav1d.so.4 (0x00007f5b66aef000)
	librsvg-2.so.2 => /lib/x86_64-linux-gnu/librsvg-2.so.2 (0x00007f5b660f1000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f5b66095000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f5b65f66000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f5b65e41000)
	libzvbi.so.0 => /lib/x86_64-linux-gnu/libzvbi.so.0 (0x00007f5b65db1000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5b65d94000)
	libsnappy.so.1 => /lib/x86_64-linux-gnu/libsnappy.so.1 (0x00007f5b65d89000)
	libaom.so.0 => /lib/x86_64-linux-gnu/libaom.so.0 (0x00007f5b658ef000)
	libcodec2.so.0.9 => /lib/x86_64-linux-gnu/libcodec2.so.0.9 (0x00007f5b64b0a000)
	libgsm.so.1 => /lib/x86_64-linux-gnu/libgsm.so.1 (0x00007f5b64afa000)
	libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f5b64a82000)
	libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f5b64a25000)
	libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007f5b649ca000)
	libshine.so.3 => /lib/x86_64-linux-gnu/libshine.so.3 (0x00007f5b647bd000)
	libspeex.so.1 => /lib/x86_64-linux-gnu/libspeex.so.1 (0x00007f5b647a2000)
	libtheoraenc.so.1 => /lib/x86_64-linux-gnu/libtheoraenc.so.1 (0x00007f5b64765000)
	libtheoradec.so.1 => /lib/x86_64-linux-gnu/libtheoradec.so.1 (0x00007f5b64745000)
	libtwolame.so.0 => /lib/x86_64-linux-gnu/libtwolame.so.0 (0x00007f5b64720000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f5b646f3000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f5b64646000)
	libwavpack.so.1 => /lib/x86_64-linux-gnu/libwavpack.so.1 (0x00007f5b6461a000)
	libx264.so.160 => /lib/x86_64-linux-gnu/libx264.so.160 (0x00007f5b64358000)
	libx265.so.192 => /lib/x86_64-linux-gnu/libx265.so.192 (0x00007f5b633db000)
	libxvidcore.so.4 => /lib/x86_64-linux-gnu/libxvidcore.so.4 (0x00007f5b632c8000)
	libva.so.2 => /lib/x86_64-linux-gnu/libva.so.2 (0x00007f5b6329b000)
	libmfx.so.1 => /lib/x86_64-linux-gnu/libmfx.so.1 (0x00007f5b6328b000)
	libxml2.so.2 => /lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f5b630dd000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f5b630ca000)
	libgme.so.0 => /lib/x86_64-linux-gnu/libgme.so.0 (0x00007f5b6307b000)
	libopenmpt.so.0 => /lib/x86_64-linux-gnu/libopenmpt.so.0 (0x00007f5b62e91000)
	libchromaprint.so.1 => /lib/x86_64-linux-gnu/libchromaprint.so.1 (0x00007f5b62e79000)
	libbluray.so.2 => /lib/x86_64-linux-gnu/libbluray.so.2 (0x00007f5b62e2e000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f5b62c2e000)
[... 73 lines cut ...]
	libthai.so.0 => /lib/x86_64-linux-gnu/libthai.so.0 (0x00007f5b60b74000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f5b60a8c000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f5b60a5b000)
[... cut ...]
	libdatrie.so.1 => /lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f5b5eacb000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f5b5ea9d000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f5b5ea96000)
```

but a newly compiled UVtools (`bash build/compile.sh && bash build/run.sh`) still complains about missing OpenCV (though no more ABI / libc complaints!)

Full build log attached.
[build-log.txt](https://github.com/sn4k3/UVtools/files/9965863/build-log.txt)

sn4k3, 2022-11-08:
Then all you need is copy `emgucv-4.6.0/libs/libcvextern.so` to my release. the build scripts are still copying the nuget opencv which run new glibc.

Download linux package in zip format (not AppImage).
Replace my with your libcvextern.so
Run UVtools.

If it run attach your file to me

sn4k3, 2022-11-08:
Also have you changed my build script? It looks like it is linking libraries we don't need not trimming opencv

Heres what i expect from compiled "mini" library:

```
	linux-vdso.so.1 (0x00007ffc62fd4000)
	libgeotiff.so.5 => /lib/x86_64-linux-gnu/libgeotiff.so.5 (0x00007f481801b000)
	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f4817ffd000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4817dd3000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4817cec000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4817ccc000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4817aa2000)
	libtiff.so.5 => /lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f4817a1c000)
	libproj.so.22 => /lib/x86_64-linux-gnu/libproj.so.22 (0x00007f48176b3000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007f4817689000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f481b637000)
	libwebp.so.7 => /lib/x86_64-linux-gnu/libwebp.so.7 (0x00007f481761d000)
	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f481754c000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f4817521000)
	libjbig.so.0 => /lib/x86_64-linux-gnu/libjbig.so.0 (0x00007f4817510000)
	libjpeg.so.8 => /lib/x86_64-linux-gnu/libjpeg.so.8 (0x00007f481748f000)
	libdeflate.so.0 => /lib/x86_64-linux-gnu/libdeflate.so.0 (0x00007f481746b000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f481744f000)
	libsqlite3.so.0 => /lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f4817302000)
	libcurl-gnutls.so.4 => /lib/x86_64-linux-gnu/libcurl-gnutls.so.4 (0x00007f481725f000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007f4817235000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007f4817214000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007f48171f5000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x00007f4817188000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x00007f4817172000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x00007f481712c000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f4816f41000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f4816eed000)
	libldap-2.5.so.0 => /lib/x86_64-linux-gnu/libldap-2.5.so.0 (0x00007f4816e8e000)
	liblber-2.5.so.0 => /lib/x86_64-linux-gnu/liblber-2.5.so.0 (0x00007f4816e7d000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f4816e6d000)
	libunistring.so.2 => /lib/x86_64-linux-gnu/libunistring.so.2 (0x00007f4816cc3000)
	libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x00007f4816c7b000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f4816bf9000)
	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007f48167b7000)
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007f481667a000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007f4816662000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f4816597000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f4816568000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f4816562000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f4816554000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007f4816537000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f4816514000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x00007f4816507000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f4816500000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f48164ec000)
```

Your is containing media libs

wohali, 2022-11-08:
Segfault, and the coredump is useless:

```
warning: Can't open file /memfd:xshmfence (deleted) during file-backed mapping note processing
[New LWP 97331]
[New LWP 97336]
[New LWP 97337]
[New LWP 97342]
[New LWP 97338]
[New LWP 97335]
[New LWP 97340]
[New LWP 97332]
[New LWP 97341]
[New LWP 97334]
[New LWP 97343]
[New LWP 97333]
[New LWP 97339]
--Type <RET> for more, q to quit, c to continue without paging--c
Core was generated by `./UVtools'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f429b2fff59 in ?? ()
[Current thread is 1 (LWP 97331)]
(gdb) bt
#0  0x00007f429b2fff59 in ?? ()
#1  0x0000557062ebc7b8 in ?? ()
#2  0x0000000000000000 in ?? ()
```

wohali, 2022-11-08:
> Also have you changed my build script? It looks like it is linking libraries we don't need not trimming opencv
> 
> Heres what i expect from compiled "mini" library:
>
> Your is containing media libs

No change to the build script, I simply cloned your repo and ran your shell script.

sn4k3, 2022-11-08:
Ok i will setup a VM to try this out, but i think  opencv updated glibc causing this trouble and linking out lib will do that seg fault

sn4k3, 2022-11-09:
I got the same result as by compiling the lib in a fresh install...

wohali, 2022-11-09:
Oh no, you mean not working? Oof. Well, I guess we won't be able to run this under Debian until 12.x releases , which is probably Summer 2023. 😢

sn4k3, 2022-11-09:
Any chance to install ubuntu or mint?

Other solution would be use my zip package with the older lib and remove permitions of libcvextern.so, keep only `x` permition. That way i guess my auto upgrade won't replace that lib. 

Meanwhile i have updated upgrade script to not install on such machines.

wohali, 2022-11-09:
I can run a VM if necessary, but it is not ideal.

Well, anyway, I can use 3.7.2 until next year (when I will file a bug to revert your upgrade script change 😉)

sn4k3, 2022-11-10:
Solved

closed 2022-11-10 by sn4k3

wohali, 2022-11-10:
Confirmed working -- thank you for your hard work on this to get it out so fast!!!