sn4k3/UVtools #607: [BUG]

VersusJH, 2022-11-18:
Hi! Thank you for this tool and thanks in advance for any help you'll be able to give me.

## System
 - OS: Arch Linux
 - Slicer: X
 - Printer: Anycubic Mono 4k
 - UVtools version: 3.8.3 Appimage

```
UVtools v3.8.3 X64
Operative system: Linux 6.0.8 #1 SMP PREEMPT_DYNAMIC Tue Nov 15 16:04:11 CET 2022 X64
Processor: AMD Ryzen 7 3700X 8-Core Processor
Processor cores: 16
Memory RAM: 7.81 / 15.55 GB
Runtime: arch-x64
Framework: .NET 6.0.11
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary)
    WA: 1920 x 1051    UA: 1920 x 1051

Path:       /tmp/.mount_uvtoolvV3B59/usr/bin/
Executable: /tmp/.mount_uvtoolvV3B59/usr/bin/UVtools
Loaded file: base.pwma [Version: 516] [Class: PhotonWorkshopFile]

```

## Describe the bug

I am trying to generate a .pwma file for my printer from a .gerber file, using the PCB exposure tool. Howver, only the upper left corner of the circuit is contained in the generate file, while the rest of the circuit is missing.

## To Reproduce

1. Open file "base.pwma" containing base settings for my printer
2. Open the PCB exposure tool and load the test.gbr file.
3. With any setting (I have not observed any differences) click "PCB exposure" to run the tool
4. Corner of PCB is shown, but not the rest.

## Expected behavior

I would expect the whole PCB to be imported - see attached screenshots.

## Screenshots

[image]

[image]

## Files

Had to zip them to upload them to github. Contains aforementioned base.pwma and test.gbr

[bugfiles.zip](https://github.com/sn4k3/UVtools/files/10039787/bugfiles.zip)

---

Thank you again, an have a good day!

sn4k3, 2022-11-18:
Nothing wrong.
You just placed your PCB far from origin (0,0).
UVtools place the board exactly on the coordinates you set in design.

[image]

On CAD move your PCB to 0,0 or near it to make it inside your printer build size. Still if it exceed LCD dimensions it will be cut

comato, 2022-11-22:
I have same issue with gerber file... "Missing parts". Gerber is generated from EasyEDA (basic version). Green traces on picture are from Kicad gerber viewer, white traces is screenshot from UVTools.
[image]
[Gerber_TopLayer.zip](https://github.com/sn4k3/UVtools/files/10061671/Gerber_TopLayer.zip)

sn4k3, 2022-11-22:
@comato can you prepare a minimal file only with that marked area?

comato, 2022-11-22:
I prepared it, same... gerpber is in ZIP
[Gerber_TopLayer.zip](https://github.com/sn4k3/UVtools/files/10065133/Gerber_TopLayer.zip)
[image]

comato, 2022-11-24:
Nice, thank you :)

sn4k3, 2022-12-01:
Fixed

closed 2022-12-01 by sn4k3