sn4k3/UVtools #609: [Bug]

comato, 2022-11-22:
### System

```Text
System.ComponentModel.Win32Exception (0x80070057): The parameter is incorrect.
   at Avalonia.Win32.SystemDialogImpl.<>c__DisplayClass2_0.<ShowFileDialogAsync>b__0() in /_/src/Windows/Avalonia.Win32/SystemDialogImpl.cs:line 120
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__272_0(Object obj)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(Thread threadPoolThread, ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot, Thread threadPoolThread)
--- End of stack trace from previous location ---
   at UVtools.WPF.Controls.Tools.ToolLayerImportControl.AddFiles() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Controls\Tools\ToolLayerImportControl.axaml.cs:line 183
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.AvaloniaSynchronizationContext.<>c__DisplayClass5_0.<Post>b__0() in /_/src/Avalonia.Base/Threading/AvaloniaSynchronizationContext.cs:line 33
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 283
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32Platform.RunLoop(CancellationToken cancellationToken) in /_/src/Windows/Avalonia.Win32/Win32Platform.cs:line 210
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\Program.cs:line 70
Category: Application

# System
UVtools v3.5.6 X64
Operative system: Microsoft Windows 6.1.7600 X64
Processor: Intel(R) Core(TM) i5-4670K CPU @ 3.40GHz
Processor cores: 4
Memory RAM: 6.25 / 15.95 GB
Runtime: win7-x64
Framework: .NET 6.0.7
AvaloniaUI: 0.10.17
OpenCV: 4.5.5-237

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe

Machine date time: 11/22/2022 01:01:50
    UTC date time: 11/22/2022 00:01:50
```

### Printer and Slicer

- Printer: Photon Mono
- Slicer: Photon Workshop

### Description of the bug

Import layers create crash

### How to reproduce

1. go to "Actions"
2. "Import layer"
3. "+Add"
4. select file in attachment

### Files

[image]

sn4k3, 2022-11-22:
Duplicate of: #590
I suspect that is caused by having dual monitor when both are ON, is that your case?

comato, 2022-11-22:
> Duplicate of: #590 I suspect that is caused by having dual monitor when both are ON, is that your case?

No, I have enabled only one 4k screen

sn4k3, 2022-11-22:
Ok, all this errors are caused by UI framework which is outside UVtools scope. I can never replicate those as it always works for me:

[image]

Does that happens everytime you attempt to do it?

sn4k3, 2022-11-22:
Also try to load file from it real path, don't use windows library to access files, because a user reported this errors happens in windows 7 libraries shortcuts but works fine on 10 and 11

comato, 2022-11-22:
> Also try to load file from it real path, don't use windows library to access files, because a user reported this errors happens in windows 7 libraries shortcuts but works fine on 10 and 11

It was reason of crash, thank you. If I use real absolute path, it is working.

But... some patterns are not imported, no new layers, still UVtools default layers, nothing happened. Only import window is automatically colsed. I created more patterns in ZIP file... 
[import issue.zip](https://github.com/sn4k3/UVtools/files/10061876/import.issue.zip)