sn4k3/UVtools #610: [Bug]

comato, 2022-11-22:
### System

```Text
UVtools v3.8.3 X64
Operative system: Microsoft Windows 6.1.7600 X64
Processor: Intel(R) Core(TM) i5-4670K CPU @ 3.40GHz
Processor cores: 4
Memory RAM: 6.52 / 15.95 GB
Runtime: win7-x64
Framework: .NET 6.0.11
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 3840 x 2160 @ 100% (Primary) (On this)
    WA: 3840 x 2120    UA: 3840 x 2120

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Photon Mono FW v0.1.7 UI_001
- Slicer: Photon Wokshop

### Description of the bug

Converted file to Photon Mono (PWMO) is not printed correctly. It is same with all versions of exported pwmo file (1, 515, 516, 517) and also same with latest and older versions of UVTools.
Preview in slicer Photon Workshop is correct, but on printer LCD is something wrong with pwmo files generated from UVtools. 
PWMO files generated from .stl in Photon Workshop are printed correctly.

### How to reproduce

1. Open UVTools
2. Export layers to Photon Mono PWMO file. You can leave default layers or edit, import new layers. It does not matter.
3. Print this PWMO file on Photon Mono printer
[image]

### Files

[photon mono issue.zip](https://github.com/sn4k3/UVtools/files/10061577/photon.mono.issue.zip)

sn4k3, 2022-11-22:
Are you trying to convert the demo file to pwmo? If so you can't do that as it contain different resolution and display size.
You need to open an valid pwmo file and edit from there.

comato, 2022-11-22:
Ahh ok. If I opened pwmo file generated from Photon Wokrshop and edit/add layers in this file, it is printed correctly :) Thank you for tip.

closed 2022-11-22 by comato