sn4k3/UVtools #611: [Bug] Issue: layers sorted lexicographically instead of numerically

bartoszek, 2022-11-25:
### System

```Text
UVtools v3.8.3 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: Intel(R) Core(TM) i7-3720QM CPU @ 2.60GHz
Processor cores: 8
Memory RAM: 7.46 / 15.58 GB
Runtime: win10-x64
Framework: .NET 6.0.11
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 114.58333333333333% (Primary) (On this)
    WA: 1920 x 1046    UA: 1676 x 913

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: xenomorph_legs_hollow_upside_down_tilted_27,5ml_12h40m.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: NA
- Slicer: NA

### Description of the bug

Layers in the Issue detector view is sorted lexicographically instead of numerically 

### How to reproduce

1. Got to the Issues view
2. Run Detect
3. Click on "Layers" column header to trigger sorting.

### Files

[image]

sn4k3, 2022-11-25:
Layer column is a string (not number) because of issues that have a range of layers.
I have no control over the sort as it comes from UI framework (DataGrid).

However UVtools sorts by default by layer index as number.
You can also change default Sort on Settings, which may solve this problem

bartoszek, 2022-11-26:
Why not change `Layers` column sorting to `programmatic` and override the `Sort` method?

sn4k3, 2022-11-26:
I will take a look at it

sn4k3, 2022-12-01:
Fixed

closed 2022-12-01 by sn4k3