sn4k3/UVtools #625: [Bug] macOS M1 arm64 - 3.9.3 - Package damaged or incomplete

mawburn, 2022-12-15:
### System

```Text
- UVTools v3.9.3 arm64
- macOS Ventura 13.0.1
  - Apple M1
```

### Printer and Slicer

n/a

### Description of the bug

Doesn't install from command line or run through the zip. 

[image]
[image]

It actually also caused an issue with the old version (3.9.2) when I tried to reinstall it, where I have to resign it through the command line. 

```cmd
codesign --force --deep --sign - ./UVtools.app
```

Not sure if it's related, but figured it was worth a mention.

### How to reproduce

Tried to install from:

```cmd
bash -c "$(curl -fsSL https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/install-uvtools.sh)"
```

And the .zip file. 

https://github.com/sn4k3/UVtools/releases/download/v3.9.3/UVtools_osx-arm64_v3.9.3.zip

sn4k3, 2022-12-15:
I have checked, and in fact the `arm64` package miss the UVtools binary, I dont know why... All others packages are fine.
I have reupload the arm64, please check.

Note: The auto-installer script does the codesign for you. No need to do manually.

mawburn, 2022-12-15:
@sn4k3 that was fast! 

I forgot to mention that I originally tried downloading from the app itself, which I just retried and it works great! Thanks!

closed 2022-12-15 by mawburn