sn4k3/UVtools #638: `PhotonWorkshopFile.Header`: Byte addresses in comments are off by 4 bytes

DominikPalo, 2023-01-03:
### System

```Text
N/A
```

### Printer and Slicer

- Printer: Anycubic Photon M3
- Slicer: Lychee Slicer 5.0.001

### Description of the bug

Not a real bug, just a documentation issue: it seems that field comments in the [`PhotonWorkshopFile.Header`](https://github.com/sn4k3/UVtools/blob/1386476b7ab50b85b8f410a5e3d5f6f616071cf9/UVtools.Core/FileFormats/PhotonWorkshopFile.cs#L251) class are off by 4 bytes. Let me explain:

Here is a code snippet:
```
        /// <summary>
        /// 40
        /// </summary>
        [FieldOrder(1)] public float PixelSizeUm { get; set; } = 47.25f;

        /// <summary>
        /// Layer height in mm
        /// 44
        /// </summary>
        [FieldOrder(2)] public float LayerHeight { get; set; }
```

According to the comments, the `PixelSizeUm` should be stored at position 0x40 and the `LayerHeight` at position 0x44. But, when I examine my `.pm3` file, the `PixelSizeUm` = 40 is stored at position 0x44, the `LayerHeight` = 0.05 is at position 0x48, and so on (every field is offset by 4 bytes):
[image]

Anyway, the file is deserialized properly in the app 
[image]

### How to reproduce

N/A

### Files

[B2 Arrow Button.pm3.zip](https://github.com/sn4k3/UVtools/files/10335476/B2.Arrow.Button.pm3.zip)

sn4k3, 2023-01-03:
With new fields introduction and re-order that addresses may be out-dated/wrong. I will remove them from comments.

DominikPalo, 2023-01-03:
@sn4k3 thank you :) BTW, let me share a funny story behind finding the issue:

My initial idea was to create a simple Python plugin for KiCad to generate .pm3 files directly from the KiCad PCB tool - I usually design PCBs in KiCad and I wanted to involve my Anycubic Photon M3 printer in the process of DIY DPS manufacturing. So I started googling to check if there is some library or code snippet describing the .pm3 format and I somehow bumped into this project (I didn't know anything about UVtools before). Then I went through the sources, especially the PhotonWorkshopFile class, to better understand the Anycubic file structure. But, what a surprise - I suddenly found out that UVTools already has similar functionality - the **PCB Exposure tool**! 😍 This is exactly what I need and it works perfectly! It is even better, as it supports Gerber files, so it's not limited only to KiCad. I'm really thankful for this great addition, good job :)

sn4k3, 2023-01-03:
Not bad at all, most users who use UVtools daily are not aware of most of it features ;)

In sources you can find (commented) code to translate KiCad files to layer images, but soon I drop that and start to parse gerber instead.

Here a sample from a member with 0.1mm paths:

[image]

closed 2023-01-03 by sn4k3