sn4k3/UVtools #640: System.AggregateException: One or more errors occurred. (Image ended short: 17323202, expecting: 20736000)

bk4-bit, 2023-01-03:
### System

```Text
UVtools v3.10.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: Intel(R) Core(TM) i7-10510U CPU @ 1.80GHz
Processor cores: 8
Memory RAM: 13.76 / 15.87 GB
Runtime: win10-x64
Framework: .NET 6.0.12
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1032    UA: 1920 x 1032
2: 1920 x 1080 @ 100%
    WA: 1920 x 1032    UA: 1920 x 1032

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Anycubic Mono X 6k
- Slicer: Chitubox, Lychee

### Description of the bug

This error apears when im trying to open any pwmb files from usb stick that was already in printer. Same files that was saved from UVtools to harddrive opens normaly. Same file that only copyed to USB and opened just after that opens normaly. Also there some errors during print with thin stripes on VAT (sometimes on all vat, sometimes partialy), i suspect that its can be connected. In file AmeraLabs_calibration_part_town was edited only "Wait before cure" parameter, and that all, there also was this strange stripes pattern in the VAT.

System.AggregateException: One or more errors occurred. (Image ended short: 17323202, expecting: 20736000)
 ---> System.IO.FileLoadException: Image ended short: 17323202, expecting: 20736000
   at UVtools.Core.FileFormats.PhotonWorkshopFile.LayerDef.DecodePW0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 855
   at UVtools.Core.FileFormats.PhotonWorkshopFile.<>c__DisplayClass204_0.<DecodeInternally>b__1(Int32 layerIndex) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1995
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.<>c__DisplayClass19_0`1.<ForWorker>b__1(RangeWorker& currentWorker, Int32 timeout, Boolean& replicationDelegateYieldedBeforeCompletion)
   at System.Threading.Tasks.TaskReplicator.Replica.Execute()
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.TaskReplicator.Run[TState](ReplicatableUserAction`1 action, ParallelOptions options, Boolean stopOnFirstFailure)
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException(ICollection exceptions, CancellationToken cancelToken, Exception otherException)
   at System.Threading.Tasks.Parallel.ForWorker[TLocal](Int32 fromInclusive, Int32 toExclusive, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Func`4 bodyWithLocal, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](TSource[] array, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at UVtools.Core.FileFormats.PhotonWorkshopFile.DecodeInternally(OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\PhotonWorkshopFile.cs:line 1984
   at UVtools.Core.FileFormats.FileFormat.Decode(String fileFullPath, FileDecodeType fileDecodeType, OperationProgress progress) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.Core\FileFormats\FileFormat.cs:line 3553
   at UVtools.WPF.MainWindow.<>c__DisplayClass110_0.<ProcessFile>b__1() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 1343

### How to reproduce

1. edit file after Slicer in UVtools
2. copy it to usb
3. remove usb from PC
4. insert to printer
5. put again USB to PC, open the file from USB or copy to harddrive and open from there
6 same file on PC that wasnt copyed to usb opens well

### Files

[image]
[image]
[image]
[AmeraLabs_calibration_part_town.zip](https://github.com/sn4k3/UVtools/files/10339203/AmeraLabs_calibration_part_town.zip)

sn4k3, 2023-01-03:
Maybe printer modify something. 

Please provide the two files:

1. Original after UVtools save
2. After insert on printer and go back to PC

This way I can compare the two files and see what changed.

bk4-bit, 2023-01-03:
That file that was attached before is "broken" from printer. Clean file is here. Also i find out that Photon Workshop crashes after im trying to resave edited in UVtools files.
[AmeraLabs_calibration_part_town_CLEAN.zip](https://github.com/sn4k3/UVtools/files/10339665/AmeraLabs_calibration_part_town_CLEAN.zip)

sn4k3, 2023-01-03:
Both files have same size 

[image]

However file content changes and it's corrupted:

[image]

I removed the error to see the result and it's clear a corruption problem on last 2 layers:

[image]

[image]

This is not UVtools problem, printer may corrupt files somehow. Try to reflash firmware.

bk4-bit, 2023-01-04:
Thanks a lot for help and fast reply. I will test your suggestions, and reply with result.

bk4-bit, 2023-01-05:
Hi. The problem was in flash drive. Formatting did not help, problem repeat after some time. I'm changed the flash drive and everithing is ok, and also stripes problem also solved. Thanks for help.

closed 2023-01-05 by bk4-bit