sn4k3/UVtools #644: [FEATURE REQUEST] Allows plugin to interact with the user in a better way

yaqwsx, 2023-01-11:
### Is your feature request related to a problem?

UVTools plugins are extremely useful. However, at the moment, there is no reasonable way (as far as I am aware) to report any debug logs or report an error that we find during image processing.

Similarly, user input must be entered before the plugin is started. Sometimes, we would like to prompt a user for a simple yes/no answer based on the image analysis.

### Describe the solution you'd like

I would like to see a simple textual log displayed below the plugin progress bar and the corresponding methods of the plugin class to fill it in.

Similarly, it would be nice to have a method to report an error (possibly throw an exception of suitable type instead) which will be displayed to the user in a reasonable matter that suggests there is no bug in the plugin, but the source data are unexpected (i.e., do not show stack trace).

sn4k3, 2023-01-11:
Scripts run in the Core which is not aware of the UI so you can't trigger UI code. However the UI communicate with scripts and it's possible to make a bridge and trigger some actions to show dialogs.

> Similarly, user input must be entered before the plugin is started. Sometimes, we would like to prompt a user for a simple yes/no answer based on the image analysis.

You mean after load a script ask a question? 
Can you provide a sample/workflow so I can better visualize the solution?

> I would like to see a simple textual log displayed below the plugin progress bar and the corresponding methods of the plugin class to fill it in.

You mean a small editable text to show more detailed text? Like a Multi line text box?
If you have an visual idea please draw.

> Similarly, it would be nice to have a method to report an error (possibly throw an exception of suitable type instead) which will be displayed to the user in a reasonable matter that suggests there is no bug in the plugin, but the source data are unexpected (i.e., do not show stack trace).

Can do, also note throw exceptions do special thing in UVtools, it does a undo of whatever operation have made on layers.
Will this work:

[image]

It's also possible to show a information message after run an operation, which can be set by defining `Operation.Tag`, I might change that to a more logic class

yaqwsx, 2023-01-12:
Thank you for the response - let me clarify.

When I want to prompt the user, it usually happens once a plugin runs (e.g., something was detected in the layers). An information message (MessageBox) and a simple Yes/No prompt (again, MessageBox) should be sufficient.

Undo on exception is expected.

Regarding text logs, I am thinking about the following UI:

[image]

sn4k3, 2023-01-12:
Is that OK?

[image]

yaqwsx, 2023-01-12:
Looks great!

sn4k3, 2023-01-13:
In which point do you need the message boxes?

`ScriptInit`
`ScriptValidate`
`ScriptExecute`

sn4k3, 2023-01-16:
Implemented. Have a look at [sample file](https://raw.githubusercontent.com/sn4k3/UVtools/master/Scripts/UVtools.ScriptSample/ScriptAdvancedDialogSample.cs)

closed 2023-01-16 by sn4k3

yaqwsx, 2023-01-16:
Thank you! I will give it a try in my next script!