sn4k3/UVtools #645: PCB gerber importing 180 degree rotation [Bug]

74HC14, 2023-01-13:
### System

```Text
UVtools v3.10.0 X64
Operative system: Darwin 19.6.0 Darwin Kernel Version 19.6.0: Tue Jun 21 21:18:39 PDT 2022; root:xnu-6153.141.66~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz
Processor cores: 4
Memory RAM: 4.93 / 6.48 GB
Runtime: osx.10.15-x64
Framework: .NET 6.0.12
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1280 x 800 @ 200% (Primary)
    WA: 1280 x 777    UA: 640 x 388

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Anycubic Mono

### Description of the bug

This particular gerber file has this bug when I import it via the PCB exposure window. I have not had this issue with other gerber files in the past. I'm not even sure what footprint or object in the gerber is causing this issue, if I knew that I could just avoid using it.

### How to reproduce

Go to PCB exposure, import my gerber, window appears describing a bug or not implemented feature.

### Files

[https://files.catbox.moe/jp6fof.zip](url)
[https://files.catbox.moe/0e5cod.zip](url)

sn4k3, 2023-01-13:
It's a not implemented feature.

Most rotations in gerber are not implemented.
Please consider to force rotation not to be used and instead let it draw the path as 0º

sn4k3, 2023-01-16:
Implemented

closed 2023-01-16 by sn4k3