sn4k3/UVtools #646: [FEATURE REQUEST] Add drill file support to PCB Exposure tool

Tyrannosaurus1234, 2023-01-14:
### Is your feature request related to a problem?

Etching away the interior "drill hole" copper of a THT pad is useful because it allows your drill bit to self-center, currently the PCB Exposure/gerber tool does not acknowledge .drl (Excellon Drill) files if you try to upload them.

### Describe the solution you'd like

An option to upload .drl files to the gerber menu, which would then plot them against the copper layer, so those drill holes don't get exposed and can later be etched.

sn4k3, 2023-01-14:
drl uses a different syntax and then it's not implemented. However you can export drills as gerber and use it as inverse layer to subtract from.
See here how: https://github.com/sn4k3/UVtools/discussions/592#discussioncomment-4021769

sn4k3, 2023-01-16:
Implemented

closed 2023-01-16 by sn4k3

Tyrannosaurus1234, 2023-01-27:
Thanks for implementing! In testing, I found 2 issues with the behavior:

- holes are drawn inverted (as copper circles)
- the invert checkbox doesn't work for drl layers

sn4k3, 2023-01-27:
It works for me.
.drl files by default subtract if you use invert it will add white pixels.

You need to check this:
[image]