sn4k3/UVtools #648: [Crash] System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Object reference not set to an instance of an object.)

Brandelmeir, 2023-01-18:
### System

```Text
UVtools v3.11.0 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 7 3700X 8-Core Processor             
Processor cores: 16
Memory RAM: 29.87 / 63.93 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
```

### Printer and Slicer

- Printer: Elegoo Saturn 2
- Slicer: Lychee

### Description of the bug

```
System.AggregateException: A Task's exception(s) were not observed either by Waiting on the Task or accessing its Exception property. As a result, the unobserved exception was rethrown by the finalizer thread. (Object reference not set to an instance of an object.)
 ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at UVtools.WPF.LayerCache.set_Layer(Layer value) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\LayerCache.cs:line 40
   at UVtools.WPF.MainWindow.ShowLayer() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.LayerPreview.cs:line 1489
   at UVtools.WPF.MainWindow.ComputeIssues(IssuesDetectionConfiguration config) in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.Issues.cs:line 687
   at UVtools.WPF.MainWindow.OnClickDetectIssues() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.Issues.cs:line 630
   at UVtools.WPF.MainWindow.<set_SelectedTabItem>b__49_0() in D:\Tiago\Dropbox\Programming\C#\UVtools\UVtools.WPF\MainWindow.axaml.cs:line 205
   --- End of inner exception stack trace ---
Category: Task
File: teej_luke.ctb  [Version: 0] [Class: CTBEncryptedFile]

Machine date time: 19/01/2023 10:08:37 AM
    UTC date time: 18/01/2023 11:08:37 PM
```

### How to reproduce

* Loaded a CTB file (exported from Lychee Pro 5.0.0)
* Detected issues
* Painted on a layer to connect an island
* Realised I needed to fix a misplaced support in Lychee, so updated and exported the file from Lychee
* approximately 15 minute to 20 minute delay in which UV tools sat open in the background
* I pressed Ctrl + F5 to refresh the file
* UVTools crashed

sn4k3, 2023-01-18:
Now if you open the file it still crashes? It looks like an odd problem, hard to replicate.

Brandelmeir, 2023-01-18:
Re-opening the file works. I understand this is probably a hard one to reproduce but thought I should report the issue anyway.

Brandelmeir, 2023-01-19:
Reviewing my workflow I realised I missed a step in what I reported - I had run the repair tool to remove all islands at 5 pixels or less.  Full sequence was:

- Loaded a CTB file (exported from Lychee Pro 5.0.0)
- Detected issues
- Removed all islands at 5 pixels or less (via "Repair layers and issues")
- Painted on a layer to connect an island
- Realised I needed to fix a misplaced support in Lychee, so updated and exported the file from Lychee
- approximately 15 minute to 20 minute delay in which UV tools sat open in the background
- I pressed Ctrl + F5 to refresh the file
- UVTools crashed

sn4k3, 2023-01-19:
Does the file had any empty layer, or different layer count?

Brandelmeir, 2023-01-19:
> Does the file had any empty layer, or different layer count?

No empty layers, and no different layer count

closed 2023-01-19 by sn4k3