sn4k3/UVtools #649: [Bug] Infill HoneyComb

Dexus, 2023-01-25:
### System

```Text
UVtools v3.11.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 9 3950X 16-Core Processor            
Processor cores: 32
Memory RAM: 16.27 / 31.92 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: _mount-orbiter-2.1.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn S
- Slicer: Chitubox

### Description of the bug

The honeycomb is not filling the complete object, if your first layer is smaller and goes bigger, the left side will not filled with the pattern.

### How to reproduce

Open a ctb or what ever
- create a infill (40px, honeycomb, 255, 10px, 72px)
- check layers

### Files

https://we.tl/t-GAIfjlNlGn

sn4k3, 2023-01-25:
Found the problem, will fix on next release

sn4k3, 2023-01-31:
Fixed

closed 2023-01-31 by sn4k3