sn4k3/UVtools #653: [Crash] System.OverflowException: Value was either too large or too small for a Decimal.

DanielSmith1239, 2023-01-29:
### System

```Text
UVtools v3.11.0 Arm64
Operative system: Darwin 22.3.0 Darwin Kernel Version 22.3.0: Thu Jan  5 20:49:43 PST 2023; root:xnu-8792.81.2~2/RELEASE_ARM64_T8103 Arm64
Processor: Apple M1
Processor cores: 8
Memory RAM: 6.93 / 12.95 GB
Runtime: osx.13-arm64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
```

### Printer and Slicer

- Printer: 
- Slicer: 

### Description of the bug

```
System.OverflowException: Value was either too large or too small for a Decimal.
   at System.Number.ThrowOverflowException(TypeCode type)
   at System.Decimal.DecCalc.VarDecFromR4(Single input, DecCalc& result)
   at System.Decimal.op_Explicit(Single value)
   at UVtools.Core.Operations.OperationCalculator.MillimetersToPixels..ctor(Size resolution, SizeF display, Decimal millimeters) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.Core/Operations/OperationCalculator.cs:line 91
   at UVtools.Core.Operations.OperationCalculator.InitWithSlicerFile() in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.Core/Operations/OperationCalculator.cs:line 57
   at UVtools.Core.Operations.Operation.set_SlicerFile(FileFormat value) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.Core/Operations/Operation.cs:line 80
   at UVtools.WPF.MainWindow.ProcessFile(String fileName, FileDecodeType fileDecodeType, UInt32 actualLayer) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/MainWindow.axaml.cs:line 1604
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority) in /_/src/Avalonia.Base/Threading/JobRunner.cs:line 37
   at Avalonia.Native.PlatformThreadingInterface.SignaledCallback.Signaled(Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 39
   at Avalonia.Native.Interop.Impl.__MicroComIAvnSignaledCallbackVTable.Signaled(Void* this, Int32 priority, Int32 priorityContainsMeaningfulValue) in /_/src/Avalonia.Native/Interop.Generated.cs:line 4336
--- End of stack trace from previous location ---
   at Avalonia.Native.PlatformThreadingInterface.RunLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Native/PlatformThreadingInterface.cs:line 90
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken) in /_/src/Avalonia.Base/Threading/Dispatcher.cs:line 65
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 120
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 209
   at UVtools.WPF.Program.Main(String[] args) in /mnt/d/Tiago/Dropbox/Programming/C#/UVtools/UVtools.WPF/Program.cs:line 183
Category: Application
File: buttons.lgs30  [Version: 0] [Class: LGSFile]

Machine date time: 1/29/2023 11:25:30 AM
    UTC date time: 1/29/2023 4:25:30 PM
```

### How to reproduce

Open sl1 file and start auto-conversion, or open lgs30 file.

sn4k3, 2023-01-29:
I cannot reproduce, can you supply the `buttons.lgs30` file?
Also, error happens on calculator tool, did you use that or other tool?

DanielSmith1239, 2023-01-30:
All I did was open the `buttons.sl1`. I installed from home-brew if that makes a difference.

Here are the files: https://we.tl/t-59iuhAreAn

Also updated printer/slicer in issue description.

sn4k3, 2023-01-30:
Found the problem. Will fix on next patch.

DanielSmith1239, 2023-01-30:
Great thanks! Out of curiosity, what was the issue?

sn4k3, 2023-01-30:
LGS don't have display width / height properties instead they have pixels per millimeter. UVtools notification system was notifying about Yppmm when any resolutionX or display width change internally making it value Infinite due a division by 0 when calculating `DisplayHeight` from `Yppmm`, and that raises the error.
The lack of setter for `Display` never notify again to sane the values

Now property change notification only trigger for its exact axis. Also instead listen for change and set ppmm to file header it is seting directly on display setter, which is set during convertions.

[image]

sn4k3, 2023-01-31:
Fixed

closed 2023-01-31 by sn4k3

samsadatidds, 2024-03-05:
I am having the same issue.  I just downloaded the UV Tool as per your you tube video. When trying to open the test file for the Phrozen Sonic Mini 4K the system crashes and this is the report:

 [Crash] System.OverflowException: Value was either too large or too small for a Decimal.
at System.Number.ThrowOverflowException(TypeCode type)
   at System.Decimal.DecCalc.VarDecFromR4(Single input, DecCalc& result)
   at System.Decimal.op_Explicit(Single value)
   at UVtools.Core.FileFormats.FileFormat.RefreshPrintParametersModifiersValues() in UVtools.Core\FileFormats\FileFormat.cs:line 4916
   at UVtools.Core.Operations.OperationEditParameters.InitWithSlicerFile() in UVtools.Core\Operations\OperationEditParameters.cs:line 163
   at UVtools.Core.Operations.Operation.set_SlicerFile(FileFormat value) in UVtools.Core\Operations\Operation.cs:line 84
   at UVtools.UI.MainWindow.ProcessFile(String fileName, FileDecodeType fileDecodeType, UInt32 actualLayer) in UVtools.UI\MainWindow.axaml.cs:line 1673
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at Avalonia.Threading.SendOrPostCallbackDispatcherOperation.InvokeCore()
   at Avalonia.Threading.DispatcherOperation.Execute()
   at Avalonia.Threading.Dispatcher.ExecuteJob(DispatcherOperation job)
   at Avalonia.Threading.Dispatcher.ExecuteJobsCore(Boolean fromExplicitBackgroundProcessingCallback)
   at Avalonia.Threading.Dispatcher.Signaled()
   at Avalonia.Win32.Win32Platform.WndProc(IntPtr hWnd, UInt32 msg, IntPtr wParam, IntPtr lParam)
   at Avalonia.Win32.Interop.UnmanagedMethods.DispatchMessage(MSG& lpmsg)
   at Avalonia.Win32.Win32DispatcherImpl.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.DispatcherFrame.Run(IControlledDispatcherImpl impl)
   at Avalonia.Threading.Dispatcher.PushFrame(DispatcherFrame frame)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime(AppBuilder builder, String[] args, ShutdownMode shutdownMode)
   at UVtools.UI.Program.Main(String[] args) in UVtools.UI\Program.cs:line 173
Category: Application
File: _Phrozen_test_Sonic_Mighty_4K.ctb  [Version: 4] [Class: ChituboxFile]

Machine date time: 3/5/2024 3:20:28 PM
    UTC date time: 3/5/2024 8:20:28 PM