sn4k3/UVtools #654: [FEATURE REQUEST] add a PAUSE button, to pause processing

ninjadude333, 2023-01-30:
### Describe the solution you'd like

sometimes the processing takes a long time, and since its a high compute process it freezes the computer,
if there was an option to pause, and later resume, that would be great.
just like in file transfers or torrent downloading etc.

sn4k3, 2023-01-30:
Can be a good addition, it can take a while to implement.

To overcome the freezes you should decrease workload, on settings you can click on "!" button to assign the optimal thread count to not freeze your machine and spare some for OS: 

[image]

ninjadude333, 2023-01-30:
hi, the thing is that i want to work full power, but sometimes I need a quick pause while in the middle :)

sn4k3, 2023-01-30:
On my machine spare 2 threads make almost no diference for UVtools while it does for System.
It may worth the try, the more cores you have the less you gain, core count is not linear with performance.

Implementing a well-made pause will require a lot of changes on code, structure and how calls are made. As it is now the possible approach is to put thread to sleep, that does not release them and you may notice a performance issue, because thread is not working but is held with it resources.

I can plan this, but do not expect to be soon.

ninjadude333, 2023-01-31:
great, ill play with the settings and try to optimize it.
thanks :)

sn4k3, 2023-02-27:
Implemented

closed 2023-02-27 by sn4k3