sn4k3/UVtools #655: [Bug]

DanielSmith1239, 2023-01-31:
### System

```Text
UVtools v3.11.1 Arm64
Operative system: Darwin 22.3.0 Darwin Kernel Version 22.3.0: Thu Jan  5 20:49:43 PST 2023; root:xnu-8792.81.2~2/RELEASE_ARM64_T8103 Arm64
Processor: Apple M1
Processor cores: 8
Memory RAM: 7.28 / 13.22 GB
Runtime: osx.13-arm64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1440 x 900 @ 200% (Primary)
    WA: 1440 x 875    UA: 720 x 438

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: buttons4-2.lgs30 [Version: 0] [Class: LGSFile]
```

### Printer and Slicer

- Printer: Longer Orange 30
- Slicer: SuperSlicer

### Description of the bug

When I convert an `sl1` file to `lgs30`, it sets "Bottom Layer Count" to 0. This affects the print and isn't just an issue with the UI.

### How to reproduce

1. Download my `sl1` file and open in UV Tools (without auto-converting).
2. Go to Tools -> Edit Print Parameters, see Bottom Layer Count is not 0.
3. Convert for Longer Orange 30 (File -> Convert To -> Longer3D -> Longer Orange 30.
4. Open converted file in UV Tools, go to Edit Print Parameters and see that Bottom Layer Count is changed to 0.

### Files

[buttons4.sl1.zip](https://github.com/sn4k3/UVtools/files/10549159/buttons4.sl1.zip)

sn4k3, 2023-02-01:
Fixed in new patch, please try

closed 2023-02-01 by sn4k3

DanielSmith1239, 2023-02-01:
Confirmed, latest patch fixes the issue. Thanks!