sn4k3/UVtools #659: [Bug] Applying "Transition Layers" Fix Removes Variable Cure Times

joehhn, 2023-02-17:
### System

```Text
UVtools v3.11.2 X64
Operative system: Microsoft Windows 10.0.19043 X64
Processor: Intel(R) Core(TM) i5-10400F CPU @ 2.90GHz
Processor cores: 12
Memory RAM: 6.80 / 15.87 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 2560 x 1440 @ 150% (Primary) (On this)
    WA: 2560 x 1380    UA: 1707 x 920

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: _untitled-Cube.ctb [Version: 4] [Class: ChituboxFile]
```

### Printer and Slicer

- Printer: Elegoo Saturn 
- Slicer: ChituBox V1.9.4
Issue can be confirmed with just UVTools and the provided slice below

### Description of the bug

**Problem**: the "Transition layers" fix assumes that all bottom layers have the same b-exposure time, and all normal layers have the same n-exposure time. It calculates exposure steps with the "typical" (file-default) value of b-exposure and n-exposure, overwriting variable cure times for every layer in the slice.

**Example**: The cube slice linked below has "Dynamic Layer Heights" applied: from layers 0-74, the layer height is 100microns, and from layers 75 to 432, the layer height is 50microns. The 100micron layers have n-exposure of 4 seconds and b-exposure of 14 seconds, while the 50 micron layers have n-exposure of 3 seconds and b-exposure of 14 seconds. Transitional layers are overwritten when UVTools generates dynamic layers, so the cube has a huge change in exposure from 14s (100micron b-exposure) to 4s (100-micron n-exposure) at the end of the bottom layers. However, when I try to apply the "Transition Layers" suggestion to recreate transitional layers, UVTools assumes all normal layers have the 50micron n-exposure of 3s, _overwriting_ the 4s cure times for the 100micron layers. In my case, this means the 100micron layers no longer get extra cure time, which is just as bad as having no transition layers at all :) Exposures are being recalculated as if all layers are 50micron layers.

**Expected behavior**: transition layers should calculate the "exposure steps" between the _final_ bottom layer and the _first_ top layer, then generate a transition without overwriting per-layer cure times. In the example above, that means:
Check last bottom layer's exposure time --> 14s
Then check first normal layer's exposure time --> 4s
Calculate the difference --> 10s
Then generate exposure steps --> 11.5s, 9.0s, 6.5s
And most importantly --> don't overwrite the rest of the model! Leave the per-layer cures as-is: 100micron layers at 4s, and the 50micron layers at 3s.

**Note**: The "Transition Layers" fix might also overwrite per-layer cure times for bottom layers. I haven't tested this, but it's a likely issue if UVTools assumes that b-layers also have a constant exposure. Bottom layers with variable cure times is an edge-case, however, so I'm not nearly as concerned about that as I am about transition layers overwriting normal layer exposures. Thanks for sharing this powerful tool with us, and please keep up the great work.

### How to reproduce

1. Load the included slice file in UVTools
2. Scale up the layer data window within the "Info" tab. Now scroll through the layers. The bottom layers (100micron) have a 14s cure time, that there are no transition layers, that the first ~70 normal layers (100micron) cure at 4s exposure, and that the remaining ~360 (50micron) layers cure at 3s exposure. All we need to do here is add some transition layers between the 14s bottom and 4s 100micron layers.
3. Switch to the "Suggestions" tab and apply "Transition Layers"
4. Switch back over to the "Info" tab. As you scroll through the bottom layers through the transition and finally the normal layers, notice that the exposure times have been overwritten. Bottom layers are still 14s, and transition layers were created, but now, all normal layers have an exposure time of 3s, including the 100micron layers that came right after the bottom layers. We still have variable layer heights, but we lost the variable layer cures, so this print will likely fail.

### Files

[DynLayers Transition Incompatibility.zip](https://github.com/sn4k3/UVtools/files/10770098/DynLayers.Transition.Incompatibility.zip)

sn4k3, 2023-02-18:
I can explain the current behavior:

Every time transition layer count changes it needs to sanitize the layer exposure.

Let's say you have 20 transition layers, then you change to 10 transition layers, the UVtools need to set back 10 layers to the normal exposure. Since it doesn't know/save previous exposure it uses the base exposure to sanitize layers. 
It also happens when you set transition from 0 because your global normal time is 3s and find's a discrepancy in time between bottoms and normal and set it to 3s.

I will try to improve the current method and use more clever inspection.

joehhn, 2023-02-18:
Thanks for the explanation! I didn't realize the dilemma you have there with reconstructing exposure times. I suppose you could offer to restore exposure times for these "sanitized" layers with a new manual-entry table under the "Transition Layers" settings. Far be it from a perfect solution, but it's certainly doable since only the "Transition Layers" suggestion has this problem. Ideally, I think UVTools would fetch exposures from the "Dynamic Layer Heights" based on layer heights, but I realize that's quite a bit more complicated to implement. Anyway, thanks for taking the time to address this :) Look forward to seeing what you come up with!

sn4k3, 2023-02-27:
Fixed

closed 2023-02-27 by sn4k3

joehhn, 2023-03-07:
Incredible. Thanks for fixing this so quickly!