sn4k3/UVtools #661: [Bug] Halot One Plus File Parameters not being accepted by printer after choosing Printer Parameters

Mygnyl, 2023-02-20:
### System

```Text
UVtools v3.11.2 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: AMD Ryzen 9 5900X 12-Core Processor            
Processor cores: 24
Memory RAM: 15.25 / 31.93 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       D:\3Dp - resin\UVtools_win-x64_v3.11.2\
Executable: D:\3Dp - resin\UVtools_win-x64_v3.11.2\UVtools.exe
Loaded file: _file_to_print_v1 - chitu.cxdlp [Version: 3] [Class: CXDLPFile]
```

### Printer and Slicer

- Printer: Halot One Plus
- Slicer: PrusaSilcer v2.6.0-alpha3

### Description of the bug

Problem: When I load a file generated from UVTools on to my printer via the flash drive, if I select File Parameters then the printer will use the file parameters I put in PrusaSlicer. If I instead select Printer Parameters, it will use the settings from my printer. However, if I try to go back and use File Parameters instead, the printer will select the Printer Parameters instead. If I reload the file from the flash drive, the issue persists. The only way I have found to be able to select and use File Parameters again is by using a .cxdlp file generated by ChituBox. Once I load that file, I can load my original file and am able to select and use File Parameters. 
Expected: I expected to be able to pick and choose whichever parameters I wanted without needing to load a file generated from a different program.

### How to reproduce

1. Generate .SL1 file with PrusaSlicer
2. Load file into UVTools
3. Send generated .CXDLP file to flash drive
4. Before inserting flash drive, change all of the Printer Parameters to their max so that the print time will be very long if chosen
5. Take flash drive to printer and load this file
6. Select File Parameters and note the estimated print time.
7. Go back to the file selection screen and select the file again, this time choosing Printer Parameters
8. Note the very long print time
9. Go back to the file selection screen and select the file again, this time choosing File Parameters
10. Note that the File Parameters print time is now the same as the Printer Parameters print time
To fix, you have to load a file generated by ChituBox (I didn't try other slicers) in order to be able to select and use File Parameters again.

I thought maybe it was something in the file header so I ran a diff check on them, which you will find attached. I'm not sure if it's the header or not that was just my first thought.

For the settings that I copied to ChituBox from PrusaSlicer, these were everything under the Print tab in settings. I don't use ChituBox so I think some of the differences might be because of that.

I don't know if this is actually a UVTools bug or not, but I figured I would start here. I have not tested this with the files from PrusaSlicer 2.5.0 yet as I assume they would be handled by UVTools in the same way.

[image]

sn4k3, 2023-02-20:
It looks like a version problem. On right you use V3 which is more recent, on UVtools you are using V2.
On PrusaSlicer edit the printer and edit **FILEVERSION_3** under notes

Mygnyl, 2023-02-20:
Thanks for the quick response! That actually solved the issue!

closed 2023-02-20 by Mygnyl

eefweenink, 2023-02-24:
Maybe I noticed the same issue (or simular) with the Halot One. 
--------
Operative system: Darwin 20.6.0 Darwin Kernel Version 20.6.0: Fri Dec 16 00:35:00 PST 2022; root:xnu-7195.141.49~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i5-4308U CPU @ 2.80GHz
Processor cores: 4
Memory RAM: 4.60 / 6.88 GB
Runtime: osx.11.0-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
____________________
PrusaSlicer 2.5.0 
Halot One C60    (V1_H2.238.2a2.238.2_C2.238.2_R2.238.2.tar 01 Feb. 2023) 

-------------------
When I print in the Halot One menu I choose:
Use filesettings. And press start. 
I had lots of misprints, (with Sirya Tech Sculpt), whith 3.4 secs nothing on the plate, with 9 secs 0,05mm it looked quite will. 
Then I did a couple of test with 0,1mm (to gain some time); 12 secs. 16 secs (yes a little overexposure)
Back to 14 -15 and underexposure. 
All that time I did not look into the Printer-printersettings, but trusted it was all in the file. 

Because a lot of people told me these numbers where far too high, I went for the printersettings and not the settings in the file. That way I did succesfully two bigger prints today. In the test with 9 secs 0,05 mm I had overexposure, in the bigger things I did, I had too big supports (because of Tech Sculpt I had 0,8 and 1mm) but that was far too much. 
Anyway, I am back on track again with my printing process. 

Looking at the Halot One, there is defenitedly something not OK. I have the idea it used a kind of mix of file + printersettings (because not all the prints I did were equal). 

So what should be done here? Also change the version setting? or is more information needed? 

Regards, Eef