sn4k3/UVtools #664: [Bug] File is damaged. Print stop!

lexucd, 2023-02-20:
### System

```Text
UVtools v3.11.2 X64
Operative system: Microsoft Windows 10.0.19045 X64
Processor: Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz
Processor cores: 8
Memory RAM: 8.39 / 15.43 GB
Runtime: win10-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files (x86)\UVtools_win-x64_v3.6.3\
Executable: C:\Program Files (x86)\UVtools_win-x64_v3.6.3\UVtools.exe
Loaded file: Shaman_Last.pmx2 [Version: 516] [Class: PhotonWorkshopFile]
```

### Printer and Slicer

- Printer:  Mono X2
- Slicer: Lychee

### Description of the bug

After i import the pmx2 files into UVTools , repair them and save them again , they become unprintable. It stops at the 1st layer before the UV lamp starts.
[image]

### How to reproduce

I have tried to print the same file without the UVtools and it started printing alright.

sn4k3, 2023-02-21:
Provide two files in a zip: 
1) Original out of lychee
2) Edited by UVtools

lexucd, 2023-02-21:
[Mouse.zip](https://github.com/sn4k3/UVtools/files/10794868/Mouse.zip)

lexucd, 2023-02-23:
Did you have the chance to look at the files?

sn4k3, 2023-02-25:
I guess file with `_copy` is from UVtools? However, I can't find any logical explanation for the problem, everything looks correct.

See if that solves: [Mouse_RightB_Full_Shaman_Sup_copy.zip](https://github.com/sn4k3/UVtools/files/10832209/Mouse_RightB_Full_Shaman_Sup_copy.zip)

closed 2023-05-14 by sn4k3