sn4k3/UVtools #667: [Bug] Halot One does not use the settings in the files made by UVTools

eefweenink, 2023-02-25:
### System

```Text
UVtools v3.11.2 X64
Operative system: Darwin 20.6.0 Darwin Kernel Version 20.6.0: Fri Dec 16 00:35:00 PST 2022; root:xnu-7195.141.49~1/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i5-4308U CPU @ 2.80GHz
Processor cores: 4
Memory RAM: 3.75 / 5.87 GB
Runtime: osx.11.0-x64
Framework: .NET 6.0.13
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1280 x 800 @ 200% (Primary)
    WA: 1280 x 800    UA: 640 x 400

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
```

### Printer and Slicer

- Printer: Halot One C 60 (latest firmware)
- Slicer:  Prusa Slicer 

### Description of the bug

After days of testing and searching I finally found that the Halot One does not use the settings in the files made by UVTools. Somehow it uses the parameters what are in the printer itself (or maybe better said, makes a mix of boths). 
On the Issues page on Github I found a (bug)message what looks very simular, maybe something with Version? 
See [https://github.com/sn4k3/UVtools/issues/661...](https://github.com/sn4k3/UVtools/issues/661?fbclid=IwAR2XNj-cwMuOUH_4UpeP5i6Rj0xGABLjmsUfAhiP1ku2ww4LMUATTA47Jvo#issuecomment-1444095667) for details. 
Is it a bug, a feature ( I don't understand) or something else?

### How to reproduce

- In Halot One load file
- Choose Use File parameters
- Check parameters and I see the printer parameters.

sn4k3, 2023-02-25:
Being handled on [Facebook](https://www.facebook.com/groups/uvtools/posts/867210894388415)

sn4k3, 2023-03-18:
Use version 3

closed 2023-03-18 by sn4k3