sn4k3/UVtools #668: [Bug]

adriangalilea, 2023-02-27:
### System

```Text
UVtools v3.12.0 X64
Operative system: Darwin 22.1.0 Darwin Kernel Version 22.1.0: Sun Oct  9 20:14:54 PDT 2022; root:xnu-8792.41.9~2/RELEASE_X86_64 X64
Processor: Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz
Processor cores: 16
Memory RAM: 13.43 / 35.01 GB
Runtime: osx.13-x64
Framework: .NET 6.0.14
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1792 x 1120 @ 200% (Primary)
    WA: 1792 x 1095    UA: 896 x 548

Path:       /Applications/UVtools.app/Contents/MacOS/
Executable: /Applications/UVtools.app/Contents/MacOS/UVtools
Loaded file: UVtools_demo_file.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer: Elegoo Saturn 2
- Slicer: PrusaSlicer 2.6.0-alpha4

### Description of the bug

Frist time trying UVtools, tried following [https://github.com/sn4k3/UVtools/wiki/Setup-PrusaSlicer](Setup PrusaSlicer)

Tried importing the profiles and when I go to prusa slicer no profile was impored, when I go back to UVtools and do it again items are greyed out, so I'm unable to do anything else I guess.
[image]

### How to reproduce

1. Go to UVTools with Prusa Slicer closed
2. Go to: Menu -> Help -> Install profiles into PrusaSlicer
3. Install profiles
4. Open prusa slicer
5. Nothing is there

sn4k3, 2023-02-27:
You are using the alpha version which use separate folder to store settings to not mess with the release version. UVtools does not export to alpha, only RC and stable

adriangalilea, 2023-02-28:
> You are using the alpha version which use separate folder to store settings to not mess with the release version. UVtools does not export to alpha, only RC and stable

I see, is there any way I can manually paste the profiles into the alpha folder?

I am working on a huge size project that will greatly benefit from 1.6(new cut and plug/dowel) and would love to put my printer to work asap.

Thank you.

sn4k3, 2023-02-28:
You can copy all profiles from stable to alpha folder.
It's located here: `C:\users\<username>\AppData\Roaming\PrusaSlicer`

adriangalilea, 2023-02-28:
> You can copy all profiles from stable to alpha folder. It's located here: `C:\users\<username>\AppData\Roaming\PrusaSlicer`

I'm on Mac and after a bit of digging I found them at:
'/Users/<username>/Library/Application Support/PrusaSlicer'

I was about to create a pull request for a slight modification on the wiki page mentioning this about the alpha on how to find the both mac/windows folders, just in case, but I'm not familiar on how to do that on a wiki and I was unable to do it, can you help?

sn4k3, 2023-02-28:
Don't worry, I will put that information or allow to set on alpha too.

closed 2023-03-16 by sn4k3