sn4k3/UVtools #673: [Bug]  Uniformation GKtwo slicing file format is only compatible with CTB(Encrypted)

awaybird, 2023-03-06:
### System

```Text
UVtools v3.12.0 X64
Operative system: Microsoft Windows 10.0.19044 X64
Processor: 12th Gen Intel(R) Core(TM) i3-12100
Processor cores: 8
Memory RAM: 15.61 / 31.59 GB
Runtime: win10-x64
Framework: .NET 6.0.14
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Sreens, resolution, working area, usable area:
1: 1920 x 1080 @ 100% (Primary) (On this)
    WA: 1920 x 1040    UA: 1920 x 1040

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: box-test.sl1 [Version: 0] [Class: SL1File]
```

### Printer and Slicer

- Printer:  UniFormation GKtwo
- Slicer:  Prusa Slicer 2.5.0+win64 , chitubox 1.9.5beta3(Coming with the machine)， Uniformation Slicer 2.4.0

### Description of the bug

When I use PrusaSlicer to slice the exported .sl1 file, the conversion function of UVtools to Chitubox -> Chitubox CTB for UniFormation Gktwo menu for V4 V3 is not recognized, and GKtwo's file manager does not display the .ctb file.
But after converting to Chitubox CTB (Encrypted) format, the file can be displayed and printed normally in GKtwo.
I don't know if it needs to change the default compatible format of GKtwom to Chitubox (Encrypted)
Under firmware 1.1.1, 1.0.7 and 1.0.6, only Chitubox CTB(Encrypted) format can be displayed and printed properly.

### How to reproduce

Put the converted .ctb file on a USB drive, only Chitubox CTB(Encrypted)  format can be displayed and printed.
[image]

[image]

### Files

PrusaSlicer .sl1 file
[AmeraLabs_Town.sl1.zip](https://github.com/sn4k3/UVtools/files/10895160/AmeraLabs_Town.sl1.zip)

Chitubox CTB for UniFormation GKtwo V4（GKtwo file manager can't display ）
[AmeraLabs_Town--GKtwo-CTBV4.zip](https://github.com/sn4k3/UVtools/files/10895180/AmeraLabs_Town--GKtwo-CTBV4.zip)

Chitubox CTB(Encrypted)--（gktwo can display and print normally）
[AmeraLabs_Town--Encrypted.zip](https://github.com/sn4k3/UVtools/files/10895187/AmeraLabs_Town--Encrypted.zip)

Uniformation Slicer 2.4.0 JXS file--(displays and prints normally)
[AmeraLabs_Town_v2_raft--jxs.zip](https://github.com/sn4k3/UVtools/files/10895210/AmeraLabs_Town_v2_raft--jxs.zip)

sn4k3, 2023-03-16:
> Chitubox CTB for UniFormation GKtwo V4（GKtwo file manager can't display ）

This was sliced directly with Chitubox?

> Uniformation Slicer 2.4.0 JXS file--(displays and prints normally)

This one still uses CTB v4 (No encrypted ctb)

awaybird, 2023-03-17:
> > Chitubox CTB for UniFormation GKtwo V4（GKtwo file manager can't display ）
> 
> This was sliced directly with Chitubox?
[AmeraLabs_calibration_part_town_Slicing with chitubox-basic.zip]
> 
> > Uniformation Slicer 2.4.0 JXS file--(displays and prints normally)
> 
> This one still uses CTB v4 (Not encrypted ctb)

“Chitubox CTB for UniFormation GKtwo V4（GKtwo file manager can't display ）” This file is converted from .sl1 file using UVtools "convert to - Chitubox - Chitubox CTB for UniFormatino GKtwo" and then selecting the V4 version. I have also tested selecting the V3 version, but the result is still the same, the file list does not show the ctb file.
Here is the file sliced directly using the chitubox basic 1.9.5 beta3 version that came with the machine. which displays and prints normally.
[AmeraLabs_calibration_part_town_Slicing with chitubox-basic.zip](https://github.com/sn4k3/UVtools/files/10998448/AmeraLabs_calibration_part_town_Slicing.with.chitubox-basic.zip)

sn4k3, 2023-03-17:
Ok, I will make a patch to address the problem. Please try to dry print this file to see if it works:

[shark_coaster_0.05mm_13m.encrypted.zip](https://github.com/sn4k3/UVtools/files/11000998/shark_coaster_0.05mm_13m.encrypted.zip)

awaybird, 2023-03-20:
shark_coaster_0.05mm_13m.encrypted.zip and  shark_coaster_0.05mm_13m.gktwo.zip. Both files display properly in the file list and print properly. I tested printing both files and both should work correctly from platform movement and exposure. Then I actually printed both files using the same resin and the first .encrypted file printed successfully. The second .gktwo file failed to print with the model half in the resin vat and half on the platform (probably a setup issue).

sn4k3, 2023-03-22:
Fixed

awaybird, 2023-03-23:
> Fixed

Sorry, I didn't notice it when I tested it a few days ago, but today I suddenly found that when converting other formats to gktwo-v4.ctb, the resin usage information is lost, but when converting to encrypted format, the resin usage information is displayed normally.
[image]
[cylinder.gktwo.zip](https://github.com/sn4k3/UVtools/files/11046186/cylinder.gktwo.zip)
[cylinder-encrypted.zip](https://github.com/sn4k3/UVtools/files/11046187/cylinder-encrypted.zip)

sn4k3, 2023-03-23:
They are two diferent file formats. Use the one that is best for you.
For default I opt for gktwo.ctb which is the CTBv4, unencrypted version which is lighter to process. 
Under your printer notes you can default to encrypted.ctb

awaybird, 2023-03-24:
I tested it, the JXS file sliced with the software that comes with the machine is able to display the resin usage normally, but after changing the suffix of jxs to ctb, the machine can't display the resin usage normally. After changing the suffix back, it can be displayed again. Maybe it's the firmware. But not much effect on printing.
[cylinder - rename-to-ctb.zip](https://github.com/sn4k3/UVtools/files/11058060/cylinder.-.rename-to-ctb.zip)
[cylinder-jxs.zip](https://github.com/sn4k3/UVtools/files/11058061/cylinder-jxs.zip)

sn4k3, 2023-03-24:
You can try to rename the gktwo.ctb to jxs, it probably will show resin usage too

awaybird, 2023-03-24:
> 

> You can try to rename the gktwo.ctb to jxs, it probably will show resin usage too

If the file is converted to gktwo.ctb using sl1 format, after changing the suffix to jxs, it also cannot display the resin usage.
[cylinder.gktwo.zip](https://github.com/sn4k3/UVtools/files/11058325/cylinder.gktwo.zip)
[cylinder.gktwo-to-jxs -.zip](https://github.com/sn4k3/UVtools/files/11058326/cylinder.gktwo-to-jxs.-.zip)
[cylinder-sl1.zip](https://github.com/sn4k3/UVtools/files/11058327/cylinder-sl1.zip)