sn4k3/UVtools #674: [Bug] The EPAX E10 8K appears as Machine Unknown in UVTools with file sliced with Lychee Slicer

CoralieMango3D, 2023-03-13:
### System

```Text
UVtools v3.12.0 X64
Operative system: Microsoft Windows 10.0.22621 X64
Processor: AMD Ryzen 7 6800H with Radeon Graphics         
Processor cores: 16
Memory RAM: 9.75 / 15.25 GB
Runtime: win10-x64
Framework: .NET 6.0.14
AvaloniaUI: 0.10.18
OpenCV: 4.6.0

Path:       C:\Program Files\UVtools\
Executable: C:\Program Files\UVtools\UVtools.exe
Loaded file: poussinpost update.ctb [Version: 0] [Class: CTBEncryptedFile]
```

### Printer and Slicer

- Printer: EPAX E10 8K
- Slicer: Lychee Slicer

### Description of the bug

When we slice a scene in Lychee Slicer for the EPAX E10 8K, the Machine name is Machine : Unknown in stead of Machine : EPAX E10 8K

### How to reproduce

1. Slice the chicken in Lychee Slicer
2. Export slices to file
3. Open UVTools
4. Open slice file
5. The Machine name is Machine : Unknown in stead of Machine : EPAX E10 8K

Could you confirm everything is ok this machine in UVTools? 
That will allow us to investigate this further.

Thank you for your help!
Coralie | Lychee Team
Mango3D

### Files

[image]
[image]

sn4k3, 2023-03-13:
Can you provide the sliced file?

CoralieMango3D, 2023-03-14:
Hi,

Here it is.
It's your good old Lychee Slicer chicken, if you need to reproduce the
slicing from the object file.

Thanks so much for having a look at it.

Best regards,
Coralie | Lychee Team
Mango 3D

sn4k3, 2023-03-14:
Replies via email ignore attachments, please upload directly to github under zip.

CoralieMango3D, 2023-03-15:
[poussinpost update.zip](https://github.com/sn4k3/UVtools/files/10976986/poussinpost.update.zip)

Hi, here's the zip of the chicken. After slicing for the EPAX E10 8K.

Thank you.

sn4k3, 2023-03-15:
The problem comes from slicer, it's not output the machine name.

[image]

CoralieMango3D, 2023-03-15:
Thank you for checking it out for us.

I'll close the ticket now. 

Take care! :)

closed 2023-03-15 by CoralieMango3D